jrl-umi3218/mc_rtc #184: Can't kill mc_rtc_ticker node with Python test controller

rohanpsingh, 2021-08-11:
I followed the instructions here to build a python test controller: https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/first-controller.html 

The controller seems to run fine but I can't kill with Ctrl+C (SIGINT). Is this expected behavior?

rohanpsingh, 2021-08-11:
Infact, I also cannot turn the head from left to right using the code given [in the tutorial](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/sources/first-controller.html).  
But perhaps this could be a problem specifically in my environment.

gergondet, 2021-08-12:
I can confirm the ctrl+c issue with a Python controller. I can still kill it with `Ctrl + \` (SIGKILL) I'll try to look into it.

> Infact, I also cannot turn the head from left to right using the code given in the tutorial.

I don't have this issue. Can you move the robot by changing some joints manually in the GUI? Can you try with e.g. `mc-rtc-magnum-ticker` from [mc_rtc-magnum](https://github.com/gergondet/mc_rtc-magnum)?

Thanks

rohanpsingh, 2021-08-12:
Hi Pierre, 
1. I cannot move the robot from the GUI (I don't see any markers)
2. The robot does not move its head in mc_rtc-magnum too

rohanpsingh, 2021-08-12:
In the sample controller, the execution never passes this line:
`self.postureTask.target({self.jointName: self.robot().qu[self.jointIndex]})`

gergondet, 2021-08-12:
> I cannot move the robot from the GUI (I don't see any markers)

You can change the posture objective in the GUI (Tasks -> posture_${robot} -> Target)

> The robot does not move its head in mc_rtc-magnum too

Do you mean using `mc-rtc-magnum-ticker` instead of `rosrun mc_rtc_ticker mc_rtc_ticker`?

rohanpsingh, 2021-08-12:
Ah I see, I was looking for some markers to move the joint.  
Moving the sliders (under Tasks -> posture_${robot} -> Target) _does NOT move the robot_ at all. 

Yes I tried `mc-rtc-magnum-ticker` and the `mc-rtc-magnum` gui instead of the ROS ticker.  
Yet, the robot does not move.

gergondet, 2021-08-12:
Ok. This would indicate that the controller is stuck. You can attach gdb to the running process to check where it is stuck.

This is still with your conda setup right? The units tests for this installation are ok? (in particular the Python related tests)

Can you confirm that:
- your ticker is linked to the conda version `ldd mc-rtc-magnum-ticker`
- mc_rtc-magnum was compiled with the version in your conda directory, you can tell this by checking:
```
❯ grep mc_rtc_DIR CMakeCache.txt
mc_rtc_DIR:PATH=/usr/local/lib/cmake/mc_rtc
```
- your Python controller is linked to the conda version, `ldd $conda/lib/mc_controller/python.so`

rohanpsingh, 2021-08-12:
OK I will use gdb to debug further.  

I had not compiled `mc_rtc-magnum` in the new conda environment, my bad. I recompiled, and the problem still exists.
- IMO, now the magnum ticker is linked properly to the conda mc_rtc. Here it the output of `ldd mc-rtc-magnum-ticker`:  
```
(mcrtc) rohan@kramer:~/src/mc_rtc_magnum/build/bin$ ldd mc-rtc-magnum-ticker 
	linux-vdso.so.1 (0x00007efd5908c000)
	/home/rohan/anaconda3/envs/mujoco-gym/lib/libGLEW.so (0x00007efd58fd4000)
	/usr/lib/nvidia-460/libGL.so (0x00007efd58990000)
	libmc_control.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_control.so.1 (0x00007efd5870d000)
	libmc_rtc_gui.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_rtc_gui.so.1 (0x00007efd584ef000)
	libmc_rbdyn.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_rbdyn.so.1 (0x00007efd581a6000)
	libmc_rtc_utils.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_rtc_utils.so.1 (0x00007efd57f41000)
	libspdlog.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libspdlog.so.1 (0x00007efd57c9e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007efd57a7f000)
	libstdc++.so.6 => /home/rohan/anaconda3/envs/mcrtc/lib/libstdc++.so.6 (0x00007efd5790b000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007efd5756d000)
	libgcc_s.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libgcc_s.so.1 (0x00007efd58f73000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007efd5717c000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007efd56f4b000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007efd56c95000)
	libmc_tasks.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_tasks.so.1 (0x00007efd56827000)
	libmc_solver.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_solver.so.1 (0x00007efd565ac000)
	libmc_observers.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_observers.so.1 (0x00007efd56386000)
	libnanomsg.so.5 => /usr/local/lib/libnanomsg.so.5 (0x00007efd56131000)
	libmc_rtc_loader.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_rtc_loader.so.1 (0x00007efd55f11000)
	libltdl.so.7 => /usr/lib/x86_64-linux-gnu/libltdl.so.7 (0x00007efd55d07000)
	libTasks.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libTasks.so.1 (0x00007efd55a5a000)
	libRBDyn.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libRBDyn.so.1 (0x00007efd557de000)
	libsch-core.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libsch-core.so.1 (0x00007efd55522000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007efd55308000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007efd55103000)
	libRBDynParsers.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libRBDynParsers.so.1 (0x00007efd54ebe000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007efd54caa000)
	libgeos-3.6.2.so => /usr/lib/x86_64-linux-gnu/libgeos-3.6.2.so (0x00007efd54911000)
	libyaml-cpp.so.0.5 => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5 (0x00007efd54697000)
	/lib64/ld-linux-x86-64.so.2 (0x00007efd58e64000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007efd5435f000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007efd5415b000)
	libmc_trajectory.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libmc_trajectory.so.1 (0x00007efd53f2e000)
	libeigen-quadprog.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libeigen-quadprog.so.1 (0x00007efd53d25000)
	libstate-observation.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libstate-observation.so.1 (0x00007efd53a95000)
	libanl.so.1 => /lib/x86_64-linux-gnu/libanl.so.1 (0x00007efd53891000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007efd53689000)
	libeigen-qld.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libeigen-qld.so.1 (0x00007efd53482000)
	libboost_timer.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_timer.so.1.65.1 (0x00007efd5327c000)
	libeigen-lssol.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libeigen-lssol.so.1 (0x00007efd53031000)
	libboost_serialization.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1 (0x00007efd52df3000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007efd52bcb000)
	libgfortran.so.4 => /usr/lib/x86_64-linux-gnu/libgfortran.so.4 (0x00007efd527ec000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007efd525e7000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007efd523e3000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007efd521dd000)
	libquadmath.so.0 => /home/rohan/anaconda3/envs/mcrtc/lib/libquadmath.so.0 (0x00007efd58f2b000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007efd51fc8000
```

- This seems fine too.
```
> gre mc_rtc_DIR CMakeCache.txt
mc_rtc_DIR:PATH=/home/rohan/anaconda3/envs/mcrtc/lib/cmake/mc_rtc
```

-  But does this look alright?? `ldd /home/rohan/anaconda3/envs/mcrtc/lib/mc_controller/python.so`:  
```
	linux-vdso.so.1 (0x00007ffd05f78000)
	/home/rohan/anaconda3/envs/mujoco-gym/lib/libGLEW.so (0x00007f047fbb0000)
	/usr/lib/nvidia-460/libGL.so (0x00007f047f5ac000)
	libpython3.6m.so.1.0 => /home/rohan/anaconda3/envs/mcrtc/lib/libpython3.6m.so.1.0 (0x00007f047f265000)
	libstdc++.so.6 => /home/rohan/anaconda3/envs/mcrtc/lib/libstdc++.so.6 (0x00007f047f0f1000)
	libgcc_s.so.1 => /home/rohan/anaconda3/envs/mcrtc/lib/libgcc_s.so.1 (0x00007f047fb9c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f047ed00000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f047eacf000)
[... cut ...]
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f047de55000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f047fa3c000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f047db1d000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f047d8f5000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f047d6f1000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f047d4eb000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f047d2d6000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f047d0ce000)
```

closed 2023-09-04 by gergondet