jrl-umi3218/mc_rtc #185: [mc_rbdyn] Specify the robot limit ratio

mmurooka, 2021-08-12:
For the issue of joint position tracking performance https://github.com/isri-aist/hrp2kai/issues/13, we needed to introduce the temporary scale ratio for velocity limit and torque limit in HRP2KAI like https://github.com/jrl-umi3218/mc_rtc/commit/8e6474ab510a38595de18da87582765feca96de7

Is there a more general way to do this in mc_rtc?
It is desirable to be able to specify the ratio for each joint instead of specifying a common ratio for all joints as in temporary change.

gergondet, 2021-08-12:
Hi @mmurooka 

[KinematicsConstraint](https://jrl-umi3218.github.io/mc_rtc/json.html#ConstraintSet/KinematicsConstraint) and [DynamicsConstraint](https://jrl-umi3218.github.io/mc_rtc/json.html#ConstraintSet/DynamicsConstraint) have a `velocityPercent` setting that defaults to 0.5 and does exactly that (scale the limits provided by the robot). There's no such setting for the torque limits.

I think relatively to the issue you are facing the most reasonable approach is to edit the limits in the RobotModule as you did or correct the model. It would also be possible to have your RobotModule code load an extra configuration file to set some of those ratios as you wish if you want to try things out quickly. With specific code it's also possible to reload limits online by removing/recreating/readding the kinematics or dynamics constraint.

mmurooka, 2021-08-13:
Thanks for reply @gergondet 

Which of the following two constraints is called (i.e., which of `tasks::qp::JointLimitsConstr` and `tasks::qp::DamperJointLimitsConstr` is generated) when the following yaml file is loaded for controller's conofiguration? (More generally, which source code file can I find the procedure for constructing constraints from yaml file configuration?)

```yaml
constraints:
  - type: kinematics
# damper attribute is not specified
```

https://github.com/jrl-umi3218/mc_rtc/blob/2767769f16da315753c3a20f0cff502f25baf004/src/mc_solver/KinematicsConstraint.cpp#L14-L63

The former constructor which generates `tasks::qp::JointLimitsConstr` does not seem to use velocity limit (`robot.vl()` and `robot.vu()`).

gergondet, 2021-08-13:
Hi @mmurooka 

Yes. Without a damper the velocity limits are not taken into account. It has other issues and it's generally not recommended to use it.

```yaml
constraints:
  - type: kinematics
# - damper: attribute is not specified
```

This will pick the non-damped constraint. The loading code is [here](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_solver/DynamicsConstraint.cpp#L122)

In general the loading code for tasks/constraints is in the associated `.cpp`. You can also search for `register_load_function(type` where `type` is the `type` entry in the YAML

mmurooka, 2021-08-13:
Thank you.

I understand that
https://gite.lirmm.fr/jrp-2019/mc_hwm_planner/-/commit/5d56a60db5ed5bae70a25828000da1b320479708
should have the same effect as change https://github.com/jrl-umi3218/mc_rtc/commit/8e6474ab510a38595de18da87582765feca96de7

If more detailed adjustments for each joint becomes necessary, I think that I should change the limit value described in the robot model as you pointed out.

I close this issue.

closed 2021-08-13 by mmurooka