jrl-umi3218/mc_rtc #186: Issue with mc_log_ui.read_log memory

jaehongk, 2021-08-17:
I am trying to read multiple .bin files in a loop using python. At some point, I will get the following error.

Traceback (most recent call last):
  File "bin_2021.py", line 61, in <module>
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 97, in read_log
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 99, in read_log
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 55, in read_flat
IOError: [Errno 24] Too many open files: '/tmp/tmpkd1ytq.flat'

I believe this has to do with the memory but would you have any suggestion to overcome this issue? Are there any ways to close the read bin file?

Thank you.

gergondet, 2021-08-17:
Hi @jaehongk 

Thanks for raising the issue, it is not a memory issue but rather a file descriptor leak (too many files are opened without being closed) due to a misuse of a Python library.

The commit referenced above should fix it.

closed 2021-08-17 by gergondet

jaehongk, 2021-08-17:
Thank you for your help. I have another error which occurs occasionally.

    log = mc_log_ui.read_log(bin_file)
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 98, in read_log
    return read_log(tmpf[1], True)
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 100, in read_log
    return read_flat(fpath, tmp)
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 56, in read_flat
    nrEntries = read_size(fd)
  File "/usr/lib/python2.7/dist-packages/mc_log_ui/mc_log_ui.py", line 43, in read_size
    return ctypes.c_size_t.from_buffer_copy(fd.read(ctypes.sizeof(ctypes.c_size_t))).value
ValueError: Buffer size too small (0 instead of at least 8 bytes)

May I know what may cause this error and if there would be any way to prevent this?

gergondet, 2021-08-18:
Hi @jaehongk 

I think this can happen (randomly and rarely) if the interface crashes. Would you mind sending me a log file that triggers the issue for you? (you can send it privately to pierre.gergondet+github@gmail.com) It would help to improve the `read_log` code so that it doesn't crash on such broken log files.

commit ff4e247b (2021-08-17): [utils] Close the file opened by mkstemp

Closes #186