jrl-umi3218/mc_rtc #190: [mc_control/fsm] State name is empty in configure method

mmurooka, 2021-08-21:
For configured states from configuration file (i.e., yaml or json files), the `name()` is not initialized in `configure` method.

For example, https://github.com/jrl-umi3218/mc_rtc/commit/31307b3176f5abc8fee39e04380f56e0b49fe8e1 is the patch to reproduce this issue. You'll get the following output. Although `name()` returns correct name from `start` method, it returns empty from `configure` method.

```
[success] name from configure: 
[success] name from start: DebugNameConfigured
```

By the way, in this case, which source code is called to initialize the state? I thought the following part was it, but here the name is set before calling configure, so this issue should not occur.
https://github.com/jrl-umi3218/mc_rtc/blob/950930733d3cd9df9171ebc7c3d2f0e20ad312e3/src/mc_control/fsm/StateFactory.cpp#L236-L240

gergondet, 2021-08-23:
The `StateFactory::create(state, ctl, configure, config)` is called whenever a state named `state` is created, the arguments have the following meaning:
- `state` the name of the state
- `ctl` the controller that is "hosting" the state
- `configure` whether the state must be configured with a specific configuration (a `("configs")(state)` configuration entry in the controller or in the `Meta`/`Parallel` state depending on who is instantiating the state.
- `config` the configuration if `configure` is true

The first step is to actually create the state (called [here](https://github.com/jrl-umi3218/mc_rtc/blob/950930733d3cd9df9171ebc7c3d2f0e20ad312e3/src/mc_control/fsm/StateFactory.cpp#L230) into [this function](https://github.com/jrl-umi3218/mc_rtc/blob/950930733d3cd9df9171ebc7c3d2f0e20ad312e3/src/mc_control/fsm/StateFactory.cpp#L245))

This will do one of two things (when the state exists):
- call the object loader if we are directly loading a C++
- call the factory function

There's two possible factory function:
- one created from [`load_with_loader`](https://github.com/jrl-umi3218/mc_rtc/blob/950930733d3cd9df9171ebc7c3d2f0e20ad312e3/src/mc_control/fsm/StateFactory.cpp#L286) this is a way to forward arguments to a C++ state create function and so-far only used for the Python state loader
- the other corresponds to the case when a state has been defined with a `base` entry, the factory function is defined [here](https://github.com/jrl-umi3218/mc_rtc/blob/950930733d3cd9df9171ebc7c3d2f0e20ad312e3/src/mc_control/fsm/StateFactory.cpp#L208), it creates the provided `base` (using `::create(base)` so this might recurse a few times until you get a `base` that is a C++ or Python state) and then configure the state with the specific configuration.

In the example you provided, you see one call to `configure_` corresponding to the factory function call and not the state configuration call so the name is empty. (`configure` is false in that example when you create `DebugNameConfigured`) you'd need a `configs: { DebugNameConfigured: {} }` in your `Meta` state configuration to hit that line 

It would require some changes to get the state with its final name before all `configure_` are called. Might I ask why you needed to have the name of the state in the configure method?

mmurooka, 2021-08-23:
> Might I ask why you needed to have the name of the state in the configure method?

`name()` returns empty in the following print message
https://gite.lirmm.fr/jrp-2019/mc_hwm_planner/-/blob/bf3931de4294f78389e979a71a90e7cf03107fa0/src/states/MotionFSM/UpdateObjectPoseState.cpp#L10

The state is configured as follows
https://gite.lirmm.fr/jrp-2019/mc_hwm_planner/-/blob/master/src/states/data/HwmUpdateObjectPose.yaml
and managed by internal `fsm::MetaState` instance
https://gite.lirmm.fr/jrp-2019/mc_hwm_planner/-/blob/master/src/MotionFSM.cpp#L26-41

> It would require some changes to get the state with its final name before all configure_ are called.

This issue only affects print messages for debugging in my use case, so it is not a high priority :)

commit 1ce04517 (2020-05-13): [doc/schemas] Use allOf for State schemas

Closes #190 (GitLab)
commit d24f7287 (2020-05-13): [doc/schemas] Support allOf property

Closes #190 (GitLab)
PR #218 (2022-01-24): Get state final name in configure
Closes #190
commit 94d8062b (2022-01-24): [fsm] State gets full name before ::configure

Closes #190
closed 2022-01-24 by gergondet