jrl-umi3218/mc_rtc #195: const mc_rtc::Configuration is changed

mmurooka, 2021-09-27:
`const mc_rtc::Configuration` variable is changed when `mc_rtc::Configuration::load` is called.

I uploaded the sample case to reproduce this problem to https://github.com/mmurooka/tmp_sources/tree/master/DebugConfig

It first loads the default configuration, and then makes the customized configurations by calling `defaultConfig.load(customConfig)`.
https://github.com/mmurooka/tmp_sources/blob/9716f465111b780958e4698e2a8b351dd647c591/DebugConfig/src/DebugConfig.cpp#L6-L21
https://github.com/mmurooka/tmp_sources/blob/9716f465111b780958e4698e2a8b351dd647c591/DebugConfig/etc/DebugConfig.in.yaml#L60-L67

But, as shown in the following controller output, `defaultConfig` is changed when `load` is called.
This is strange because `defaultConfig` is defined as `const` and the argument of `load` is also `const` reference.
https://github.com/jrl-umi3218/mc_rtc/blob/b2fe81b4f418c8251d85d3ceee974c0ba7e0610a/src/mc_rtc/Configuration.cpp#L615

```bash
[info] defaultConfig: {
    "var1": -1.0,
    "var2": -1.0
}
[info] ## Setup customConfig1 ##
[info] defaultConfig: { # <- defaultConfig is changed, which is unexpected behavior
    "var2": -1.0,
    "var1": 1.0
}
[info] customConfig1: {
    "var2": -1.0,
    "var1": 1.0
}
[info] ## Setup customConfig2 ##
[info] defaultConfig: { # <- defaultConfig is changed, which is unexpected behavior
    "var1": 1.0,
    "var2": 2.0
}
[info] customConfig2: {
    "var1": 1.0,
    "var2": 2.0
}
# Here, customConfig2 is expected as follows if defaultConfig is correct
# [info] customConfig2: {
#     "var1": -1.0,
#     "var2": 2.0
# }
```

gergondet, 2021-09-27:
Hi @mmurooka 

That is because `mc_rtc::Configuration` does shallow copy, so this [line](https://github.com/mmurooka/tmp_sources/blob/9716f465111b780958e4698e2a8b351dd647c591/DebugConfig/src/DebugConfig.cpp#L12) is the same JSON object under the hood*. Hence, any modification you do to `customConfig1` is also done to `defaultConfig`

*This allows the very common pattern:
```cpp
auto sub = config("sub");
```

If you don't want a shallow copy you can do the following:
```cpp
// New object, has a different JSON object under the hood
mc_rtc::Configuration customConfig1;
customConfig1.load(defaultConfig);
```

mmurooka, 2021-09-27:
Thank you for your explanation!
https://github.com/mmurooka/tmp_sources/commit/ba4bef850bc6856b3f3a9d0eb89c31cbe4bc6a70 worked.

closed 2021-09-27 by mmurooka