jrl-umi3218/mc_rtc #204: build_and_install.sh script cannot install all components in /usr/local/

TsuruMasato, 2021-11-15:
(This issue is not so high priority. Just a small question.)

Hello, this is Tsuru.

When I was cleaning-up my environment, I have a question in build_and_install.sh script.
That script cannot run with default config values.
It stops when installing any components into "/usr/local" because of authentication. (it requires sudo, but using the script with sudo is not recommended.)

This time, I changed the config and installed them in ~/openrtp directory.

We have already known that mc_rtc is supposed to be installed at "~/openrtp", when going with Choreonoid environment.

But still, I am very curious why we don't use /usr/local repository and using /openrtp instead.
"/usr/local" is very easy for cmake users to do find_packages() as a default location.

Why are we using "~/openrtp" now?
And it may be a small problem that the default params of "INSTALL_PREFIX" is set as "/usr/local", which cannot run correctly.

arntanguy, 2021-11-15:
> That script cannot run with default config values.
It stops when installing any components into "/usr/local" because of authentication. (it requires sudo, but using the script with sudo is not recommended.)

What's not recommended is to run the script as root, e.g `sudo ./build_and_install.sh` as then all the commands would be executed with root privileges (and not just the safe ones that need it). You should run the script as a normal user `./build_and_install.sh`, and it will then ask for password for any command that requires sudo. If that's not the case for you then this is strange and the issue comes from something else (maybe you built as root once before?). Please provide the installation log file ;)

> We have already known that mc_rtc is supposed to be installed at "~/openrtp", when going with Choreonoid environment.

Not really, `mc_rtc` can be installed anywhere.

> But still, I am very curious why we don't use /usr/local repository and using /openrtp instead.
"/usr/local" is very easy for cmake users to do find_packages() as a default location.

The default is `/usr/local`. 
When installed into a non-standard directory, you can add it to `CMAKE_PREFIX_PATH` environment variable so that `find_package` can find it.

> Why are we using "~/openrtp" now?

We are not ;)

TsuruMasato, 2021-11-15:
Thank you for replying so quickly.

> and it will then ask for password for any command that requires sudo.
For me, it did not happen. So I got so confused and I made this page.
You mean, general users can directly use "build_and_install.sh" even with its original configs. right?

As you pointed out, because I had done `sudo ./build_and_install.sh` once before, that must be making some errors.
Maybe I will change the ownership related to mc_rtc components or delete them first.

Second, when I installed mc_rtc series into a manually-specified directory, I have to write CMAKE_PREFIX_PATH every time, right?
For me, it's bit annoying and maybe so easy to forget.
So I will install mc_rtc series into /usr/local as far as possible.

Thank you.

closed 2021-11-15 by TsuruMasato

arntanguy, 2021-11-15:
> You mean, general users can directly use "build_and_install.sh" even with its original configs. right?

Yes.

> As you pointed out, because I had done sudo ./build_and_install.sh once before, that must be making some errors.
Maybe I will change the ownership related to mc_rtc components or delete them first.

In that case this should solve your issue:
```
sudo chown -R tsuru src
```
Where `src` is the folder where you have mc_rtc and all its dependencies and `tsuru` is your username.

> Second, when I installed mc_rtc series into a manually-specified directory, I have to write CMAKE_PREFIX_PATH every time, right?

You can put it in your bashrc. If the script is not already suggesting to do so, then we should add it ;)

mmurooka, 2021-11-15:
> Second, when I installed mc_rtc series into a manually-specified directory, I have to write CMAKE_PREFIX_PATH every time, right?

In my case, I set `INSTALL_PREFIX` as `/home/mmurooka/jrl/install` in my build_and_install_user_config.sh
https://github.com/jrl-umi3218/mc_rtc/blob/b60b42d7d6da821c7fc0320162a476c472b076e7/utils/build_and_install_user_config.sample.sh#L22

And I add the following lines to `~/.bashrc`, and it is working fine for a year :)
```bash
export PATH=~/jrl/install/bin:$PATH
export LD_LIBRARY_PATH=~/jrl/install/lib:$LD_LIBRARY_PATH
export PKG_CONFIG_PATH=~/jrl/install/lib/pkgconfig:$PKG_CONFIG_PATH
export PYTHONPATH=~/jrl/install/lib/python2.7/site-packages:$PYTHONPATH
export CMAKE_PREFIX_PATH=~/jrl/install:$CMAKE_PREFIX_PATH
```

Maybe these configurations are not suggested in scripts or tutorials.

commit 7bd414ac (2020-03-04): [mc_control] Fix an issue with MCController::loadRobot

1. Make it part of the public interface so it can be invoked by FSM
states
2. Call solver().updateNrVars() in case no constraints is added to the
robot right away

Fix #204