jrl-umi3218/mc_rtc #205: Error when linking mc_old_bin_to_flat (when cross-compiling mc_rtc for QNX)

magnusmusngi, 2021-11-16:
Hi,

Thanks for the advice in my previous [question](https://github.com/jrl-umi3218/mc_rtc/issues/201) about cross-compiling dependencies for mc_rtc. All dependencies are now cross-compiled.

However, now, as I try to build mc_rtc, I get an error while linking mc_old_bin_to_flat:

```
[ 94%] Linking CXX executable mc_old_bin_to_flat
cd /home/dteam/port_to_qnx/mc_rtc/build/utils && /usr/bin/cmake -E cmake_link_script CMakeFiles/mc_old_bin_to_flat.dir/link.txt --verbose=1
/home/dteam/qnx710/host/linux/x86_64/usr/bin/q++ -Vgcc_ntox86_64 -lang-c++  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  CMakeFiles/mc_old_bin_to_flat.dir/mc_old_bin_to_flat.cpp.o -o mc_old_bin_to_flat   -L/home/dteam/port_to_qnx/f2c/libf2c  -Wl,-rpath,/home/dteam/port_to_qnx/mc_rtc/build/src:/home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib:/home/dteam/port_to_qnx/cross-compile/eigen-quadprog/lib:/home/dteam/port_to_qnx/cross-compile/state-observation/lib:/home/dteam/port_to_qnx/cross-compile/libtool-2.4.6/lib:/home/dteam/port_to_qnx/cross-compile/mc_rbdyn_urdf/lib:/home/dteam/port_to_qnx/cross-compile/RBDyn/lib:/home/dteam/port_to_qnx/cross-compile/Tasks/lib:/home/dteam/port_to_qnx/cross-compile/sch-core/lib:/home/dteam/port_to_qnx/cross-compile/eigen-qld/lib:/home/dteam/port_to_qnx/f2c/libf2c:/home/dteam/port_to_qnx/cross-compile/geos-3.9.1/lib:/home/dteam/port_to_qnx/cross-compile/nanomsg/lib: ../src/libmc_control.so.1.8.2 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_serialization.so.1.77.0 ../src/libmc_tasks.so.1.8.2 ../src/libmc_trajectory.so.1.8.2 /home/dteam/port_to_qnx/cross-compile/eigen-quadprog/lib/libeigen-quadprog.so.1.1.1 /home/dteam/port_to_qnx/f2c/libf2c/libf2c.a /home/dteam/port_to_qnx/cross-compile/state-observation/lib/libstate-observation.so.1.4.1 ../src/libmc_solver.so.1.8.2 ../src/libmc_observers.so.1.8.2 ../src/libmc_rtc_gui.so.1.8.2 ../src/libmc_rbdyn.so.1.8.2 ../src/libmc_rtc_loader.so.1.8.2 /home/dteam/port_to_qnx/cross-compile/libtool-2.4.6/lib/libltdl.so ../src/libmc_rtc_utils.so.1.8.2 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_filesystem.so.1.77.0 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_atomic.so.1.77.0 /home/dteam/port_to_qnx/cross-compile/spdlog/lib/libspdlog.a /home/dteam/port_to_qnx/cross-compile/mc_rbdyn_urdf/lib/libmc_rbdyn_urdf.so.1.1.0 /home/dteam/port_to_qnx/cross-compile/RBDyn/lib/libRBDynParsers.so.1.5.2 /home/dteam/port_to_qnx/cross-compile/yaml-cpp/lib/libyaml-cpp.a /home/dteam/port_to_qnx/cross-compile/tinyxml2/lib/libtinyxml2.a /home/dteam/port_to_qnx/cross-compile/Tasks/lib/libTasks.so.1.5.0 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_timer.so.1.77.0 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_chrono.so.1.77.0 /home/dteam/port_to_qnx/cross-compile/RBDyn/lib/libRBDyn.so.1.5.2 /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0 /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_system.so /home/dteam/port_to_qnx/cross-compile/boost_1_77_0/lib/libboost_serialization.so /home/dteam/port_to_qnx/cross-compile/eigen-qld/lib/libeigen-qld.so.1.2.1 -lf2c /home/dteam/port_to_qnx/cross-compile/geos-3.9.1/lib/libgeos.so.3.9.1 /home/dteam/port_to_qnx/cross-compile/nanomsg/lib/libnanomsg.so.5.1.0 -lsocket 
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::__1::char_traits<char> >::~basic_binary_oprimitive()'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_iprimitive<boost::archive::binary_iarchive, char, std::__1::char_traits<char> >::~basic_binary_iprimitive()'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_iprimitive<boost::archive::binary_iarchive, char, std::__1::char_traits<char> >::init()'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::__1::char_traits<char> >::basic_binary_oprimitive(std::__1::basic_streambuf<char, std::__1::char_traits<char> >&, bool)'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_iprimitive<boost::archive::binary_iarchive, char, std::__1::char_traits<char> >::basic_binary_iprimitive(std::__1::basic_streambuf<char, std::__1::char_traits<char> >&, bool)'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::__1::char_traits<char> >::init()'
/home/dteam/qnx710/host/linux/x86_64/usr/bin/x86_64-pc-nto-qnx7.1.0-ld: /home/dteam/port_to_qnx/cross-compile/sch-core/lib/libsch-core.so.1.2.0: undefined reference to `boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::__1::char_traits<char> >::save(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
utils/CMakeFiles/mc_old_bin_to_flat.dir/build.make:131: recipe for target 'utils/mc_old_bin_to_flat' failed
make[2]: *** [utils/mc_old_bin_to_flat] Error 1
make[2]: Leaving directory '/home/dteam/port_to_qnx/mc_rtc/build'
CMakeFiles/Makefile2:2842: recipe for target 'utils/CMakeFiles/mc_old_bin_to_flat.dir/all' failed
make[1]: *** [utils/CMakeFiles/mc_old_bin_to_flat.dir/all] Error 2
make[1]: Leaving directory '/home/dteam/port_to_qnx/mc_rtc/build'
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```
`boost::archive` is part of _Boost::serialization_, which was already added as a project dependency (in _CMakeLists.txt_) and whose shared library was also already linked (in _src/CMakeLists.txt_) by default in _sch-core_.

I already tried the [suggestion](https://github.com/boostorg/serialization/issues/164) of adding `add_definitions("-D_GLIBCXX_USE_CXX11_ABI=0")` in the root CMakeLists.txt (I added it in sch-core; and in another test, I added it to my toolchain file), and removing the lines `-fvisibility=hidden` and `-fvisibility-inlines-hidden` in _libs/serialization/build/Jamfile.v2_ when building Boost. However, they both didn't worked.

Thank you very much

gergondet, 2021-11-17:
Hi @magnusmusngi 

You can check which symbols are available in the library. If you find symbols that are very similar but seem to refer the CXX11 ABI then you might be on the right track, for example, on Ubuntu 18.04 I get this symbol for the last one in your error message:

```bash
>  nm -ADC /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1|grep binary_oarchive|grep "::save("
# ... removing the overloads for bool, const char *, const wchar_t * and std::wstring, 
/usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1:000000000001ada0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
```

If  you have these symbols in your boost library as well then it means it is in fact built with the C++11 ABI so adding `-D_GLIBCXX_USE_CXX11_ABI=0` in sch-core will not solve the problem but setting to 1 should. However, it's probably a better idea to set the definitions in your toolchain file because this setting should be the same across every libraries that are going to pass `std::string` to each other.

I'm going to close #201 since it seems we are getting closer to a result here :)

magnusmusngi, 2021-11-19:
Thanks @gergondet 

Upon looking at the symbols of boost serialization's library file and comparing it to the ones in the error message, I noticed that their difference was actually in the addition of `__1::` in some of the function parameters used by sch-core. These `__1::` weren't present in the library's symbols (see sample below). I learned that this was caused by the use of LLVM C++ as the C++ library (which was actually used by QNX by default), whereas, in the jam file that I use for building Boost, I had it explicitly set to use the GNU C++ library.
```
000000000001dcc0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save_binary(void const*, unsigned long)
000000000001d940 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::init()
000000000001d290 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(bool)
000000000001d6d0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(char const*)
000000000001d800 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(wchar_t const*)
000000000001d340 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
000000000001d5a0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::save(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&)
000000000001d280 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::This()
000000000001e0d0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::basic_binary_oprimitive(std::basic_streambuf<char, std::char_traits<char> >&, bool)
000000000001e0d0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::basic_binary_oprimitive(std::basic_streambuf<char, std::char_traits<char> >&, bool)
000000000001dba0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::~basic_binary_oprimitive()
000000000001dba0 W boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >::~basic_binary_oprimitive()
0000000000240668 V typeinfo for boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >
0000000000031ca0 V typeinfo name for boost::archive::basic_binary_oprimitive<boost::archive::binary_oarchive, char, std::char_traits<char> >
```
I rebuilt Boost with LLVM C++, and the undefined references related to boost::archive methods got solved (and reached 100% when building, but not yet successful). However, at the end, I have a new set of undefined reference errors with `boost::program_options`, and still with a `__1::` issue.

After spending some time trying to fix the issue, I then just tried to cross-compile mc_rtc and all of its dependencies using GNU C++ as the C++ library, and then mc_rtc got built successfully.

By the way, I also added -`D_GLIBCXX_USE_CXX11_ABI=1` in my toolchain file after seeing `__cxx11::` in the boost library symbols.

gergondet, 2021-11-22:
Thanks for the detailed feedback.

> After spending some time trying to fix the issue, I then just tried to cross-compile mc_rtc and all of its dependencies using GNU C++ as the C++ library, and then mc_rtc got built successfully.

Then I guess we can close this issue. Feel free to re-open if you face more issues.

commit a876d0e3 (2020-05-06): [doc] Add a page to help debugging LSSOL output 6

Closes #205 (GitLab)
closed 2021-11-22 by gergondet