jrl-umi3218/mc_rtc #211: "velocityPercent" in Constraints

Taherifar, 2021-12-10:
I have found that "velocityPercent" in DynamicsConstraint.cpp and KinematicsConstraint.h has been set to 0.5 and it was advised to keep it 0.5. When I keep this value to 0.5, half of the velocity limit set in the URDF file under "velocity" is considered in IK. 
Is that possible to change this value to 1.0 instead of multiplying the URDF file velocity by two? Is there any side effect (It might be related to the way that limits are applied on IK)? Why it is advised to set "velocityPercent" to 0.5?

gergondet, 2021-12-13:
Hi @Taherifar 

> Is that possible to change this value to 1.0 instead of multiplying the URDF file velocity by two?

Yes. There's two ways you can change this value depending on your usage:
- If you are creating the constraint from C++/Python code then it's the fourth argument in the [constructor](https://jrl-umi3218.github.io/mc_rtc/doxygen.html#structmc__solver_1_1DynamicsConstraint). Actually if you build it "manually" this way the velocity percentage is 1.0 already
- if you are using yaml to load the constrain then you can use the [`velocityPercent` field](https://jrl-umi3218.github.io/mc_rtc/json.html#ConstraintSet/DynamicsConstraint), e.g.:
```yaml
constraints:
- type: dynamics
  robot: jvrc1
  damper: [0.1, 0.05, 0.5]
  velocityPercent: 1.0
```

> Is there any side effect (It might be related to the way that limits are applied on IK)? Why it is advised to set "velocityPercent" to 0.5?

This has no side effect (besides changing the velocity limits of course). I think 0.5 is an "historical" limit. When the robot is controlled via the network it might happen that a control frame is missed and the robots' safety could be triggered if that happens during a fast motion. By setting the limit to 0.5 you buy yourself one iteration that you can miss "safely". However, if mc_rtc is not sending commands over the network I don't think there is much need for setting this 0.5 value although depending on the underlying hardware and how sensitive the safety system is, you might want to set a velocity percent limit of 0.95 or so

Taherifar, 2021-12-14:
Thanks for your complete annswer. I will close this issue.

closed 2021-12-14 by Taherifar