jrl-umi3218/mc_rtc #214: Log file update timing

mmurooka, 2021-12-21:
I'm using `mc_rtc::Logger` as follows in the project which is not related to mc_rtc controller.
```
std::make_shared<mc_rtc::Logger> logger;
logger = std::make_shared<mc_rtc::Logger>(mc_rtc::Logger::Policy::THREADED, "/tmp", "my-name");
MC_RTC_LOG_HELPER("my_var", my_var_);
logger->log();
```
Link to the codes (internal only):
https://github.com/isri-aist/optmotiongen/blob/dcedbc70eb1234923c507c3366666f5feba1c1b9/optmotiongen/src/Problem/Problem.cpp#L23

I am solving a relatively large optimization problem in the project, and for each optimization iteration, I call `logger->log()`. One iteration may take up to 10 seconds, in which case `logger->log()` will be called only once every 10 seconds. I found that the bin file generated in `/tmp` is zero bytes for a while (for example, until the end of the first 20 iterations, that is around 200 sec for my case), and then becomes non-zero bytes when the program is terminated. While the bin file is zero bytes, `mc_log_ui` cannot visualize the log data. This is inconvenient, and I would like to visualize the log data at any time by updating the bin file after each iteration.

The behavior was same even when policy is `mc_rtc::Logger::Policy::NON_THREADED`.

gergondet, 2021-12-21:
Hi @mmurooka 

you can call the flush method to force a disk flush of the log. This is only implemented with the non threaded policy

mmurooka, 2021-12-21:
@gergondet Thank you, I confirmed it works!

commit 6f7b1d95 (2020-05-07): Merge branch 'topic/BetterGripperInterface' into 'master'

Better gripper interface

Closes #214, #210, and #211

See merge request multi-contact/mc_rtc!211
closed 2021-12-21 by mmurooka