jrl-umi3218/mc_rtc #215: VectorTestUnit (Failed) when installation

RakugenSon, 2021-12-28:
I tried to install mc_rtc with building from source and met the following error. 
The log file was attahced.

1: Test timeout computed to be: 1500
1: Running 5 test cases...
1: ../tests/VectorTest.cpp(225): error: in "MotionVecdLeftOperatorsTest": check crossM.vector() == crossMVecRes.col(0) has failed [ 7823.13
1: -2754.31
1: -6517.97
1:  513.896
1: -3485.99
1: -6561.86 !=  7823.13
1: -2754.31
1: -6517.97
1:  513.896
1: -3485.99
1: -6561.86]
1: ../tests/VectorTest.cpp(237): error: in "MotionVecdLeftOperatorsTest": check crossF.vector() == crossFVecRes.col(0) has failed [ 6610.27
1:  4034.06
1: -2687.51
1:   1940.7
1:  3353.97
1: -2697.15 !=  6610.27
1:  4034.06
1: -2687.51
1:   1940.7
1:  3353.97
1: -2697.15]
1: 
1: *** 2 failures are detected in the test module "MotionVecd ForceVecd test"
1: 

The following tests FAILED:
	  1 - VectorTestUnit (Failed)
Errors while running CTest
-- [ERROR] Testing is still failing for jrl-umi3218/SpaceVecAlg, please investigate deeper
Installation failed.

[build_and_install_warnings-2021-12-27-12-01-04.log](https://github.com/jrl-umi3218/mc_rtc/files/7782128/build_and_install_warnings-2021-12-27-12-01-04.log)

gergondet, 2021-12-28:
Hi @RakugenSon 

That's surprising. From the logs I can see you are running Ubuntu 20.04 with the default compiler and those tests should not be failing (they are run regularly in mc_rtc CI) so I'm not sure why it's failing on your system and how I could reproduce the issue.

You can skip testing for this build by using the `--build-testing false` option

RakugenSon, 2021-12-28:
Thanks for your reply, after I using `--build-testing false`, new error happened.
[build_and_install_warnings-2021-12-28-10-18-54.log](https://github.com/jrl-umi3218/mc_rtc/files/7782224/build_and_install_warnings-2021-12-28-10-18-54.log)

gergondet, 2021-12-28:
Oh I think I get it, it looks like you are compiling on an ARM computer?

For a quick fix you can comment out this line: https://github.com/jrl-umi3218/Tasks/blob/master/CMakeLists.txt#L25 and this one as well https://github.com/jrl-umi3218/mc_rtc/blob/master/src/CMakeLists.txt#L15

PR #217 (2022-01-11): Improve aarch64 support, remove loader sandboxing and honor gnu install dirs
This PR introduces a number of changes to improve the build on aarch64:

- Check that `-msee2` is supported before enabling the flag
- Honor GNUInstallDirs
- Remove `ObjectLoader<T>` sandboxing support. It has been disabled by default for a while and never worked well in the first place

Closes #215
closed 2022-01-11 by gergondet