jrl-umi3218/mc_rtc #216: execution cycle logging

fkanehiro, 2022-01-05:
Is there a way to log execution cycle of controller?

gergondet, 2022-01-06:
Hi @fkanehiro 

Do you mean the time between two calls to `MCGlobalController::run()`?

If so, I usually do it at the outer level, i.e. instead of:
```cpp
while(true)
{
  updateSensors();
  gc.run();
  updateCommand();
}
```

You can have:
```cpp
using clock = std::chrono::high_resolution_clock;
auto prev_start_t = clock::now()
std::chrono::duration<double, std::milli> loop_dt{0};
gc.controller().addLogEntry("perf_LoopDt", [&]() { return loop_dt.count(); });
while(true)
{
  auto start_t = clock::now();
  loop_dt = start_t - prev_start_t;
  prev_start_t = start_t;
  updateSensors();
  gc.run();
  updateCommand();
}
```

fkanehiro, 2022-01-06:
Thank you @gergondet !

Do you think it is okay to implement it in `mc_openrtm` permanently?

gergondet, 2022-01-06:
Yes, I'll close this in favor of the discussion in jrl-umi3218/mc_openrtm#16

closed 2022-01-06 by gergondet