jrl-umi3218/mc_rtc #219: build_and_install.sh cannot pass the build process

TsuruMasato, 2022-01-17:
Today, I tried to update mc_rtc and related libraries by using "build_and_install.sh".
However, after pulling all components, it causes an error in building process.

It seems to be happening at mc_rtc python binding.
If I had known much about mc_rtc, I can investigate the source code by myself...

Could you check the below images?

Sorry.

[image]
[image]

gergondet, 2022-01-17:
Hi @TsuruMasato 

Is this on Ubuntu 16.04?

gergondet, 2022-01-17:
@fkanehiro the script uninstalls projects before building to avoid this kind of issues in the bindings build

The reason I'm asking for Ubuntu 16.04 is that I stopped supporting it a while back as I though there was no more active user on this distribution and it's a little harder to keep things working (but it should be fine without the Python bindings)

fkanehiro, 2022-01-17:
> @fkanehiro the script uninstalls projects before building to avoid this kind of issues in the bindings build

Oh, really?
I must have mistaken it for another matter.

TsuruMasato, 2022-01-18:
Thank you so much for your response, Dr.@gergondet .
(Sorry for my late reply, I had a meeting this morning.)

Unfortunately, Ubuntu version is 18.04.6 LTS.
According to Kanehiro-sensei's comments, I will continue trying the script.
But if you have any idea, could you give me?

Thank you.

TsuruMasato, 2022-01-18:
I don't know why, but it works well now ...

First, I tried building without Python bindings (edited user_config.sh), and it passed.
After that, I tried with Python bindings option again, and it worked well.

The attached picture is the corresponding part of previous error.
The same warning message is still seen, but now it cam pass.

I have no ideas why this issue happened, and I cannot re-produce it now...
Anyway, it looks very rare case, so let me close this issue here.

Sorry for disturbing you.
 
[image]

closed 2022-01-18 by TsuruMasato

gergondet, 2022-01-18:
It might be related to the issue mentioned by @fkanehiro although the script has a few workarounds to handle incremental builds nicely. I'm glad it's fixed now and sorry for the trouble.

commit 323184ab (2020-05-12): [utils] mc_rtc_new_fsm_controller specifies robots by name

Closes #219

- Handle robots by name instead of indices following !214 MR
- Fixes a minor bug where FSM contact loading did not throw if contacts
were specified but invalid, and silently replaced the whole contact set
with an empty set. E.g the following entry (missing `r1`) would not report an error,
and an empty contact list would end up being used instead:

```
contacts:
- r2: ground
  r1Surface: LeftFoot
  r2Surface: AllGround
```

The expected behaviour here is to throw an error "r1 entry does not
exist", as the user explicitely  [... cut]