jrl-umi3218/mc_rtc #221: Make build_and_install script more versatile

mehdi-benallegue, 2022-01-26:
We need more and more to share frameworks with different projects that don't belong to the standard set (such as lipm walkins, footstep planner, mc state observation, etc.) so installing, updating and rebuilding the framework becomes tedious and requires to prepare different installation scripts. However I think that build_and_install scripts has nice features that we would like to use for these frameworks. 

Is it possible to make this script more versatile by allowing it to read a configuration file containing

- Pre-script commands (e.g. drcutil scripts)
- A list of additional apt dependencies.
- A list of additional packages (all to be installed after the common packages). Each with 
 * address of the repo and the corresponding branch
 * additional cmake options 
 * cutom commands to be executed before or after 
- Same for additional catkin ws
- Post-script commands
- Text to be displayed at the end (e.g. add to bashrc ...).

Maybe I got something completely wrong but I believe it would be useful.

gergondet, 2022-01-26:
Hi @mehdi-benallegue 

I have actually faced the same issue recently (i.e. setup projects where "Get mc_rtc installed" is step 1) and I have worked on a solution called [mc-rtc-superbuild](https://github.com/mc-rtc/mc-rtc-superbuild) (naming inspired by [robotology-superbuild](https://github.com/robotology/robotology-superbuild))

It can do almost everything that build_and_install does. The two current exceptions are uninstalling before updating - which I will consider to reintroduce as an option - and telling you what to put in your .bashrc which will be done by generating scripts that can be sourced to simplify that setup.

Regarding your list of desired features...

> Pre-script commands (e.g. drcutil scripts)

Possible by providing empty `CONFIGURE_COMMAND` and `INSTALL_COMMAND` and a `BUILD_COMMAND` that runs the script for example:

```
AddProject(drcutil
  GITHUB jenkinshrg/drcutils
  GIT_TAG jenkins
  CONFIGURE_COMMAND "cp config.sh.sample config.sh"
  BUILD_COMMAND "./getsource.sh && ./setupenv.sh && ./install.sh"
  INSTALL_COMMAND ""
  SKIP_TEST
)
```

(or using [FetchContent](https://cmake.org/cmake/help/latest/module/FetchContent.html)) to do this part at configuration time

(or convert drcutil content to a mc-rtc-superbuild extension)

> A list of additional apt dependencies

```
# no-op if the packages are already installed
install_apt_dependencies(libfoo-dev libbar-dev)
```

> A list of additional packages (all to be installed after the common packages)

This is exactly what this was designed for :) The only bit missing might be about custom commands to be executed before or after, if you have something a little more specific in mind we can see how to support it

> Same for additional catkin ws

```
CreateCatkinWorkspace(${SOURCE_DIRECTORY}/my_new_ws)
AddCatkinProject(my-project
  GITHUB my-org/my-project
  GIT_TAG main
  WORKSPACE ${SOURCE_DIRECTORY}/my_new_ws
  DEPENDS mc_rtc my_awesome_lib
)
```

> Post-script commands

Also a little unclear what you would need here but we can work something out also

> Text to be displayed at the end (e.g. add to bashrc ...)

I think that could fit in the environment generation thing

Extra features:
- you can compile multiple projects in parallel (extra useful with the CLONE_ONLY option)
- you can lock part of the updates
- you can easily switch the remote and/or the branch of any project

mehdi-benallegue, 2022-01-27:
Wow, looks promising I am glad that we did not start to design our own solution! 

I need to investigate this solution and let you know if anything is missing (I would an issue/PR there)

Three questions:
- Is it currently ready to be used with a framework such as ana avatar?
- Is the order of installations defined by a dependency graph. And in such a case, can we make mc_rtc itself dependent on an additional custom package?
- Is the build_and_install about to be deprecated?

gergondet, 2022-01-28:
> Is it currently ready to be used with a framework such as ana avatar?

I think so. The trickiest part is probably to integrate the drcutil script in the mix.

> Is the order of installations defined by a dependency graph. And in such a case, can we make mc_rtc itself dependent on an additional custom package?

I think the following should work:
```cmake
ExternalProject_Add_StepDependencies(mc_rtc configure my-project)
```

> Is the build_and_install about to be deprecated?

Yes. But it will be maintained for a while as there's no way to migrate from a build_and_install setup to mc-rtc-superbuild without rebuilding everything

commit e8de9b0a (2020-05-28): [RobotByName] Handle LookAtSurfaceTask

- Deprecates use of surfaceRobotIndex in favor of surfaceRobot
- Makes surfaceRobot optional (defaults to MainRobot)

Closes #221
closed 2023-09-04 by gergondet