jrl-umi3218/mc_rtc #232: dimWeight and activeJoints in task configuration

mmurooka, 2022-03-04:
If `dimWeight` and `activeJoints/unactiveJoints` are specified at the same time in the task configuration in the YAML file, `dimWeight` is ignored.
I confirmed this with the following `MomentumTask`, but I think it is also true for other tasks that inherit from `TrajectoryTaskGeneric`.

```yaml
MomentumTask:
    type: momentum
    stiffness: 100.0
    weight: 1000.0
    dimWeight: [1, 1, 1, 0, 0, 0]
    activeJoints: [ <Joint names...> ]
```

I think this is because `dimWeight` is not taken over in the following line when setting active joints.
https://github.com/jrl-umi3218/mc_rtc/blob/ea030def79f099f9b179e6eea0feef9cc569677a/include/mc_tasks/TrajectoryTaskGeneric.hpp#L216-L219

closed 2022-03-07 by gergondet

gergondet, 2022-03-07:
Thanks for reporting the issue. It has been fixed in the commit referenced above.

mmurooka, 2022-03-07:
@gergondet Thank you!

commit 46095304 (2022-03-07): [mc_tasks] Conserve dimWeight when joints selector is changed

Closes #232