jrl-umi3218/mc_rtc #233: Issue using mc_openrtm: chorenoid not showing any robot

maximebrunet, 2022-03-04:
Hello guys,

I allow myself my post from `mc_openrtm` as I don't know if you look at issues there. (https://github.com/jrl-umi3218/mc_openrtm/issues/4#issuecomment-1057793284)

I am trying to lauch chorenoid from a fresh `mc_openrtm` install (in a clean ubuntu 18 vm), but I see no robot in the chorenoid window, 
and errors in the message window.

I am doing `choreonoid sim_mc.cnoid | tee /tmp/chorenoid.log`, and I get in the chorenoid console

```
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidBalancerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidOpenRTMPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidOpenHRP3.1Plugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidGRobotPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidCorbaPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidRobotAccessPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidGrxUIPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidHrpsys31Plugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.7/libCnoidBodyPlugin.so"
Body-plugin has been activated.
GRobot-plugin has been activated.
SimulationExecutionContext has been registered.
SimulationPeriodicExecutionContext has been registered.
ChoreonoidExecutionContext has been registered.
ChoreonoidPeriodicExecutionContext has been registered.
OpenRTM-plugin has been activated.
PoseSeq-plugin has been activated.
Python-plugin has been activated.
PythonSimScript-plugin has been activated.
RobotAccess-plugin has been activated.
Balancer-plugin has been activated.
A NameService doesn't exist at "corbaloc:iiop:localhost:2809/NameService".
Corba-plugin has been activated.
GrxUI-plugin has been activated.
Hrpsys31-plugin has been activated.
OpenHRP3.1-plugin has been activated.
Loading project file "sim_mc.cnoid" ...
Task sequencer '' has been deactivated.
PythonPlugin: "/usr/lib/python2.7/dist-packages/hrpsys" has been added to the Python module search path list.
Restoring ExtCommandItem "ModelLoader"
External command "openhrp-model-loader" has been executed by item "ModelLoader".
Restoring WorldItem "World"
Restoring BodyItem "JVRC1"
Loading Body "/usr/share/OpenHRP-3.1/robot/JVRC1/main.wrl"
illegal field event type at line 103 of /usr/share/OpenHRP-3.1/robot/JVRC1/main.wrl
 -> failed.
Error: "JVRC1" cannot be restored.
Restoring BodyItem "longfloor"
Loading Body "/usr/share/OpenHRP-3.1/sample/model/longfloor.wrl"
illegal vector element at line 95 of /usr/share/OpenHRP-3.1/sample/model/longfloor.wrl
 -> failed.
Error: "longfloor" cannot be restored.
Restoring AISTSimulatorItem "AISTSimulator"
Restoring PythonSimScriptItem "sim_mc.py"
Loading Python Script for Simulation "/usr/share/hrpsys/samples/JVRC1/sim_mc.py"
 -> ok!
4 / 6 item(s) have been loaded.
Warning: 2 item(s) were not loaded.
Project "sim_mc.cnoid" has been partially loaded.
OpenGL version is 3.1.
GLSL version is 1.40.
```
I looked at the `chorenoid.log` with `PARANOID` log level, but there is not much error info ther according to me.

Could you help me fix this please ? This is preventing me from using `mc_openrtm` at all !

Thanks a lot,
Maxime

gergondet, 2022-03-07:
Hi @maximebrunet 

Could you check the `locale` inside your VM? If it's a locale where the decimal separator is not `.` but `,` (e.g. `fr_FR.UTF-8`) this might explain the issue.

In any case, can you try running:
```
export LC_ALL=C
export LANG=C
./clear-omninames.sh
choreonoid sim_mc.cnoid
```

Thanks

maximebrunet, 2022-03-07:
That was it (`LC_ALL=C` especially), thanks a lot ! :)

closed 2022-03-07 by maximebrunet