jrl-umi3218/mc_rtc #244: Implement in RobotModule in Python + load XML robot models

rohanpsingh, 2022-03-30:
I have two questions:

- Is it possible to implement a RobotModule in Python? In the [tutorial](https://jrl-umi3218.github.io/mc_rtc/tutorials/advanced/new-robot.html), it shows a C++ example only, so I assume it's not straightforward to do so in Python. Still, I would like to know how it could be done.

- I want to build a parser for XML models so I'd be able to something like`init(rbd::parsers::from_xml_file(xml_path))` while creating the robot module. A, do you think it is a good idea? and, B, do you have any pointers?

gergondet, 2022-03-31:
Hi Rohan,

> Is it possible to implement a RobotModule in Python?

No but it isn't impossible to implement either.

> Still, I would like to know how it could be done.

You need two pieces:
- a Python object that the user can creates that contains a `RobotModule` C++ object, we already have this in the bindings ([ref](https://github.com/jrl-umi3218/mc_rtc/blob/master/binding/python/mc_rbdyn/mc_rbdyn.pyx#L303)) but the implementation does not allow you to overwrite every member in `RobotModule`
- a loader from Python to C++. Since `RobotModule` does not implement any behavior I think the simplest approach is to have a `Python` module similar to the  [`Python` controller](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_control/python/python_controller.cpp) but it will just instantiate a Python `RobotModule` object so that you can extract the C++ `RobotModule` from inside.

You would use the loader as such:
```yaml
MainRobot: Python#my_module.MyRobot
```
Where `MyRobot` is a Python class that derives from `mc_rbdyn.RobotModule` that you instantiate in the `Python` RobotModule (and then extract the C++ RobotModule from the object)

> do you think it is a good idea?

I would avoid using the name `from_xml_file` since it's a little too broad and you probably have a particular kind of XML model in mind (so e.g. `from_mujoco_xml_file`?)

> do you have any pointers?

Not really. The only thing the function has to do is return a [`rbd::parsers::ParserResult`](https://github.com/jrl-umi3218/RBDyn/blob/master/src/parsers/RBDyn/parsers/common.h#L140) and from there most of the work is done :)

gergondet, 2022-07-08:
Hi @rohanpsingh 

Is this still an active topic for you?

rohanpsingh, 2022-07-08:
Hi @gergondet 
Sorry I got involved with several other stuff recently so did not have time to work on this.
I started implementing your suggestions above and ultimately want to get this done, but it's currently a low priority for me.