jrl-umi3218/mc_rtc #261: Feedforward term of AdmittanceTask

Saeed-Mansouri, 2022-06-10:
Hi,

I investigated the following commands of `AdmittanceTask.cpp` are not correct. I think by adding the feedforward term and sending it to the filter in the next iteration, it works as an integrator.

 ```
// Filter
refVelB_ = velFilterGain_ * refVelB_ + (1 - velFilterGain_) * sva::MotionVecd(angularVel, linearVel);

// Apply feed forward term
 refVelB_ += feedforwardVelB_;
```

mehdi-benallegue, 2022-06-15:
I think you are right !

More exactly, `refVelB_` would behave as a the sum of a first order filter of `(angularVel, linearVel)` with gain `(1 - velFilterGain_)*dt` and an integrator of `feedforwardVelB_` with gain `dt`. And this is definitely wrong. 

To fix it, there are two ways: (Please correct me if you think I am wrong)

- Either you really want to filter the feedforward. Is such a case the feedforward term should be multiplied with `(1 - velFilterGain_)` or, more simply added to `angularVel, linearVel` before filtering it. 
- Or you don't want to filter the feedforward term (I think it is probably what the developers wanted to do). And then filter only `angularVel, linearVel` and not the whole  `refVelB_`, then add `feedforwardVelB_`

Nevertheless, I think the effect of this error should not lead to instability in most cases, but just to a different equilibrium point (the new equilibrium point is when  `(angularVel, linearVel)==1/(1 - velFilterGain_)*feedforwardVelB_`  instead of being `(angularVel, linearVel)==feedforwardVelB_`, which may be very different according to the values of `velFilterGain_ `and `feedforwardVelB_`

Thank you for point this out.

closed 2022-07-08 by gergondet

mehdi-benallegue, 2022-07-08:
For information #272 takes solution 2.

PR #272 (2022-07-08): [mc_tasks::AdmittanceTask] Avoid filtering of feedforward velocity
Closes #261

@mehdi-benallegue can you confirm I understood your point correctly? Thanks!
commit d1df43bc (2022-07-08): [mc_tasks::AdmittanceTask] Avoid filtering of feedforward velocity

Closes #261