jrl-umi3218/mc_rtc #263: Communication between Ros and mc_rtc

Aimiliospet, 2022-06-17:
Hello everyone, 
I want to control my robot using the mc_rtc but in the robot (reemc) runs a ros controller. For that reason i want to send joint states and sensor data to mc_rtc, so that the controlling can be done there. I tried to use mc_udp to make a bridge with ros and transmit data, but i had some problems in the installation, because of the required hrpsys-base package, which followingly needs the openHRP package which i couldn't install at all. For that reason i wanted to ask if there is an alternative to mc_udp to transmit data from ros to mc_rtc. Thanks in advance!
Best regards

gergondet, 2022-06-17:
Hello @Aimiliospet 

You might be interested in https://github.com/mc-rtc/mc_rtc_ros_control although so far it has only been used with UR robots

If you want to go te mc_udp way, you can also pass `-DBUILD_OPENRTM_SERVER=OFF` when building mc_udp to disable the dependency on hrpsys-base

Aimiliospet, 2022-06-17:
Hello @gergondet,
thank you for the answer. The mc_rtc_ros_control node looks really interesting ill definetely try it, 
concerning the mc_udp now when i do the "cmake ../ -DBUILD_OPENRTM_SERVER=OFF" step i get some cmake errors for the cython bindings and uninstall. First it complains that it cant find the files in the cmake folder because these dont get cloned when i clone the repository from github, but even when i copy all the cmake files in the cmake folder i get following errors:
CMake Error at cmake/cython/cython.cmake:191 (add_dependencies):
Cannot add target-level dependencies to non-existent target "uninstall"

Call Stack (most recent call first):
  cmake/cython/cython.cmake:300 (_ADD_CYTHON_BINDINGS_TARGETS)
  binding/python/CMakeLists.txt:5 (add_cython_bindings)

Is there something big I'm missing?

kheddar, 2022-06-17:
since ReemC is a PAL humanoid, can the mc_rtc (mc_talos) developped recently by @gergondet + saeid be a good fit?

gergondet, 2022-06-17:
@Aimiliospet 

For mc_udp, you can also pass `-DPYTHON_BINDING=OFF` to disable the bindings build. To get all files from the start either:
- clone using `git clone --recursive`
- go into the source folder and run `git submodule update --init`

@kheddar 

I am not familiar with the software stack on ReemC, afaik Talos uses a modified ROS control stack for which we have something working with Saeid but there's still some work in progress. If the same stack is used on ReemC then this is definitely a good fit.

Aimiliospet, 2022-06-17:
@gergondet 
thank you very much mc_udp is now built and installed. I will also look into the other options you mentioned. Thank you for the good feedback!

closed 2023-09-04 by gergondet