jrl-umi3218/mc_rtc #264: Joint Torque Calculation

Taherifar, 2022-06-23:
Hello,

I am writing to ask how I can access the joint torques of the real robot. I have tried to create an object of `rbd::ForwardDynamics` 
 and `computeC` and `computeH` matrices, but I could not calculate the torques. Could you please help me in this regard?
(Note: I have studied [issue31](https://github.com/jrl-umi3218/mc_rtc/issues/31) and I do not want to add any dynamic constraint to the system.)

gergondet, 2022-06-24:
Hello @Taherifar 

If there is no contacts, you can do the following:

```cpp
rbd::ForwardDynamics fd{robot().mb()};
fd.computeH(robot().mb(), robot().mbc());
fd.computeC(robot().mb(), robot().mbc());
Eigen::VectorXd alphaD = rbd::dofToVector(robot().mb(), robot().mbc().alphaD);
Eigen::VectorXd tau = fd.H() * alphaD + fd.C();
```

If there is contacts, it does not make much sense not to include the dynamics constraint because the computed contact forces will be zero.

Taherifar, 2022-06-24:
Hello [gergondet](https://github.com/gergondet)

Thank you for your response. I have tried this way and it worked unless the `alphaD` is zero for all joints. I added  `robot().forwardAcceleration()` but still it was zero. I have also tried adding` fd.forwardDynamics(robot.mb(),robot.mbc())`, which results in non-zero alphaD, but the `tau` became zero. Actually `fd.H() * alphaD` became the negate of `fd.C()`.
1- Could you please help me in the above issue.
2- The robot that I am working on is a biped robot similar to JVRC and it has contact with the ground. So, if I would like to add a dynamic constraint to the real robot (not the control Robot) contact, which type of constraint should I use, and how can I input the measured wrench to the system?

Taherifar, 2022-07-06:
Thanks for your reply. So, as I run the lines of codes that you sent, I get zero values for `alphaD`, which cause the inertia term (`fd.H() * alphaD`) to become zero. The Coriolis and gravity term (`fd.C()`) has a non-zero value which has a good match with the manual calculation results. It means that If I can have a correct and non-zero value for `alphaD`, I could have the inertia and Coriolis terms and then I can add the wrench from the contacts to them and get the torque valuses. So, the question is why  `alphaD` is zero? (Note that I am using the realRobot)

gergondet, 2022-07-06:
I don't think there's an observer that updates `alphaD` for the real robot. You can try to:
- use the control `alphaD` (desired acceleration giving you desired torques)
- compute the real `alphaD` by numeric differentation

commit 648de1bb (2020-11-27): [utils/mc_log_ui] Change the default position of the labels legend

- Make the position editable
- Make the fontsize editable

Closes #264 (GitLab)
closed 2023-09-04 by gergondet