jrl-umi3218/mc_rtc #265: Build from source installation error

aamorel, 2022-06-29:
I have a problem when building from source when using the _build_and_install_default_config.sh_.

### My system

> Distributor ID:	Ubuntu
> Description:	Ubuntu 20.04.4 LTS
> Release:	20.04
> Codename:	focal

> gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

### Script output

[build_and_install_warnings-2022-06-29-11-21-35.log](https://github.com/jrl-umi3218/mc_rtc/files/9009255/build_and_install_warnings-2022-06-29-11-21-35.log)

Please tell me if you need any further information,
Thank you for your help

arntanguy, 2022-06-29:
The issue is here:
```
cd /data/aamorel/Eigen3ToPython/build && /usr/bin/cmake -E chdir /data/aamorel/Eigen3ToPython/build/python/RelWithDebInfo python -m pip install . --upgrade
/usr/bin/python: No module named pip
```

It seems that for some reason you do not have python's `pip` installed (`python3 -m pip -V`). This is surprising as the script normally tries to install it. Could you try:

```
sudo apt install python3-pip
```

Then make sure `pip` was properly installed and relaunch the `./build_and_install.sh` script.

aamorel, 2022-06-29:
I think the problem is that I have both Python 2.7 and Python 3.8 installed. 

> python -m pip
> /usr/bin/python: No module named pip

> python3 -m pip -V
> pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)

What would be the right way to tell the install to use _python3_ instead of _python_ ? Can I specify Python's path ? 
It way be the option _PYTHON_FORCE_PYTHON3_ but I wan't to be sure. 

Thank you for your fast response

aamorel, 2022-06-29:
Thank you, installation was a success !

closed 2022-06-29 by aamorel