jrl-umi3218/mc_rtc #267: No module named 'mc_log_ui'

anastasiabolotnikova, 2022-07-01:
Hello

After a fresh installation of latest version of `mc_rtc` with `PYTHON_FORCE_PYTHON3="true"` I try to run `mc_log_ui` and get an error 

```ModuleNotFoundError: No module named 'mc_log_ui```

OS: Ubuntu 20.04

Could you please recommend how to overcome it? Please let me know if any additional information is required from my side to understand the issue. Thank you very much in advance!

gergondet, 2022-07-05:
Hello @anastasiabolotnikova 

This is because (mistakenly) we don't honor the `PYTHON_FORCE_PYTHON{N}` option for mc_log_ui and it is using a separate option: `MC_LOG_UI_PYTHON_EXECUTABLE` so you can pass `-DMC_LOG_UI_PYTHON_EXECUTABLE=python3` when configuring mc_rtc and make sure you have `/usr/local/lib/python3/dist-packages` (or something corresponding to your install prefix) in your `PYTHONPATH`

anastasiabolotnikova, 2022-07-05:
Hi @gergondet 

Thank you very much for your reply. Thanks to your explanations I could resolve the problem and can now use `mc_log_ui`.

Should I close the issue or you would prefer to keep it open for now until it is addressed by honoring the `PYTHON_FORCE_PYTHON{N}` option for `mc_log_ui` ?

gergondet, 2022-07-05:
Hi @anastasiabolotnikova 

I've opened #268 to track this so I will close this one for now.

Thanks

closed 2022-07-05 by gergondet