jrl-umi3218/mc_rtc #270: ModuleNotFoundError: No module named 'mc_observers'

NourhanKhatab, 2022-07-06:
Hello,

So I'm using Ubuntu 20, ROS Neotic, and Python 3.

I have created a basic file: mctest.py

This is the file contents:

```
#!/usr/bin/env python

"""
Test for MC RTC
"""

import mc_control
import mc_rbdyn
import mc_rtc

def main():
        mc_gc = mc_control.MCGlobalController("")
        mc_robot = mc_gc.robot()
        rm = mc_robot.module()
        print("Running MC RTC controller")
        mc_gc.running = True

if __name__ == "__main__":
    main()
```
This is how I tried to run it:
`python3 mctest.py`

And this is the terminal output:
```
Traceback (most recent call last):
  File "mctest.py", line 16, in <module>
    import mc_control
  File "/usr/lib/python3/dist-packages/mc_control/__init__.py", line 1, in <module>
    from . import fsm
  File "/usr/lib/python3/dist-packages/mc_control/fsm/__init__.py", line 1, in <module>
    from .fsm import *
  File "mc_control/fsm/fsm.pyx", line 1, in init mc_control.fsm.fsm
  File "mc_observers/mc_observers.pxd", line 7, in init mc_control.mc_control
  File "/usr/lib/python3/dist-packages/mc_observers/__init__.py", line 1, in <module>
    from .mc_observers import *
ModuleNotFoundError: No module named 'mc_observers.mc_observers'
```
Thank you

@hu-yue

gergondet, 2022-07-07:
Hello @NourhanKhatab 

This is an issue with the packaging of the Python bindings. I will push a fix during the day and a new stable release to address this in the stable packages as well.

Thanks

closed 2022-07-08 by gergondet

gergondet, 2022-07-08:
Hello @NourhanKhatab 

This is now fixed on the head mirror. I will publish a release within the day to have it on the stable mirror as well.

NourhanKhatab, 2022-07-08:
@gergondet thanks for the fix, I installed the head version to test it out.
But now I am getting this error:
```
Traceback (most recent call last):
  File "mctest.py", line 16, in <module>
    import mc_control
  File "/usr/lib/python3/dist-packages/mc_control/__init__.py", line 1, in <module>
    from . import fsm
  File "/usr/lib/python3/dist-packages/mc_control/fsm/__init__.py", line 1, in <module>
    from .fsm import *
ImportError: /lib/x86_64-linux-gnu/libmc_solver.so.1: undefined symbol: _ZN5tasks2qp15CollisionConstr12addCollisionERKSt6vectorIN3rbd9MultiBodyESaIS4_EEiiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPN3sch8S_ObjectERKN3sva10PTransformIdEEiSG_SJ_SO_ddddRKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEEST_
```

gergondet, 2022-07-11:
Hi @NourhanKhatab 

Can you try to update/upgrade the packages?

I think you caught an unfortunate glitch in the system where packages for Tasks were updated but mc_rtc packages were not yet. I have tried just now in a clean docker for Ubuntu 20 + ROS noetic + Python 3 and I have no such issue.

commit 8239ca10 (2022-07-08): [debian] Include mc_observers in the Python bindings

Closes #270