jrl-umi3218/mc_rtc #281: Import mc_log_ui from Python scripts in CI

mmurooka, 2022-08-11:
When the Python scripts import `mc_log_ui` in GitHub Actions, the following error occurred.
https://github.com/mmurooka/BaselineWalkingController/runs/7788070524?check_suite_focus=true

The last error message is
```bash
ImportError: Cannot load backend 'Qt5Agg' which requires the 'qt5' interactive framework, as 'headless' is currently running
```
This probably comes from the point that CI does not have GUI. ref: https://stackoverflow.com/a/59919355
Can we prevent this error from occurring?

FYI
I'm trying to load the mc_rtc logs in GitHub Actions after the robot walks on Choreonoid to make sure the robot doesn't fall over.
https://github.com/isri-aist/BaselineWalkingController/blob/master/.github/workflows/scripts/checkSimulationResults.py
https://github.com/isri-aist/BaselineWalkingController/blob/7ea88e7af7f163d7d93b53eb8f007f5219ed0cf1/.github/workflows/ci.yaml#L126-L172

closed 2022-08-30 by gergondet

mmurooka, 2022-08-30:
Thank you @gergondet It worked on CI.

Could you tell me how often the stable apt repo for mc_rtc is updated?

gergondet, 2022-09-01:
> Could you tell me how often the stable apt repo for mc_rtc is updated?

Anytime there is a new release for mc_rtc or any of its dependencies (which are under our control) whereas the head repository any time there is a push to the master/main branch of mc_rtc or any of its dependencies

commit df3422b9 (2022-08-30): [mc_log_ui] Avoid failed imports in headless environments

Fixes #281