jrl-umi3218/mc_rtc #284: init_pos in controller configuration file does not work

mmurooka, 2022-09-02:
I added the following line to reproduce [this](https://github.com/jrl-umi3218/mc_rtc/issues/177) but it seems not working.
```yaml
init_pos:
  translation: [20, 0.0, 0.8275]
```
I assume that the robot is displayed away from the origin on Rviz, but it is displayed at the origin.
I tried it in several kinds of controllers and it didn't seem to work in any of them.

There is the following message in controller output. 
```bash
[warning] [KinematicInertial] Real anchor frame jumped from [   20.0621 0.00402126 0.00279068] to [ 0.0639299 0.00400111 -0.0651317] (error norm 0.000 > threshold 0.020)
```

arntanguy, 2022-09-02:
There indeed seems to be issues. I just tried on a simple controller, and:

In choreonoid:
- the real robot pose gets set to `init_pos`
- the control robot pose remains the default one from the robot module

In ticker: it works.

mmurooka, 2022-09-02:
> From a quick trial of LIPMStabilizer controller with both JVRC1 and HRP2KAI (using the Attitude observer), I could not reproduce https://github.com/jrl-umi3218/mc_rtc/issues/177

That is different from the behavior I have tried in my local environment.
With putting control/real robot to 20m forward from the origin with `init_pos` thanks to https://github.com/jrl-umi3218/mc_openrtm/pull/21, JVRC1 falls down forward after around 10 secs in choreonoid with lipm_walking_controller. And https://github.com/jrl-umi3218/mc_rtc/pull/285 solved it.

gergondet, 2022-09-05:
Thanks @mmurooka for the report and @arntanguy for the quick fix!

closed 2022-09-05 by gergondet