jrl-umi3218/mc_rtc #288: Ubuntu 20.04 build_and_install.sh has python issues for sva

wyqsnddd, 2022-09-07:
Hello Pierre! 

I am configuring mc_rtc on an HP machine with Ubuntu 20.04. The `build_and_install.sh` generated the error message below. Could help me with it please? 

```sh
-- Building git dependency jrl-umi3218/SpaceVecAlg (with test)
--> Compiling SpaceVecAlg (branch master)
+ cmake /home/yuquan/local/SpaceVecAlg -DCMAKE_INSTALL_PREFIX:STRING=/usr/local -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=ON -DPYTHON_BINDING:BOOL=ON -DPYTHON_BINDING_USER_INSTALL:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON2:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON3:BOOL=OFF -DPYTHON_BINDING_BUILD_PYTHON2_AND_PYTHON3:BOOL=OFF -DMC_LOG_UI_PYTHON_EXECUTABLE:STRING=python3 -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -GNinja -DBUILD_PYTHON_INTERFACE:BOOL=OFF (pwd: /home/yuquan/local/SpaceVecAlg/build)
-- Doxygen rendering: using LaTeX backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yuquan/local/SpaceVecAlg/build
+ cmake --build . --config RelWithDebInfo (pwd: /home/yuquan/local/SpaceVecAlg/build)
[1/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/include/sva_wrapper.hpp /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/include/sva_wrapper.hpp
[2/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/sva/c_sva_private.pxd /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/c_sva_private.pxd
[3/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/sva/sva.pxd /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/sva.pxd
[4/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/tests/test_sva_pickle.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/tests/test_sva_pickle.py
[5/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/sva/c_sva.pxd /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/c_sva.pxd
[6/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/tests/test_sva_vector.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/tests/test_sva_vector.py
[7/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/tests/test_sva_inertia.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/tests/test_sva_inertia.py
[8/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/sva/sva.pyx /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/sva.pyx
[9/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/tests/test_sva_containers.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/tests/test_sva_containers.py
[10/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/tests/test_sva_ptransform.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/tests/test_sva_ptransform.py
[11/12] cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E copy_if_different /home/yuquan/local/SpaceVecAlg/binding/python/sva/__init__.py /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/__init__.py
[12/12] Generating local sva python bindings
FAILED: binding/python/CMakeFiles/sva-python-bindings 
cd /home/yuquan/local/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E chdir /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo python setup.py build_ext --inplace
/usr/lib/python2.7/dist-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /home/yuquan/local/SpaceVecAlg/build/binding/python/sva/python/RelWithDebInfo/sva/sva.pxd
  tree = Parsing.p_module(s, pxd, full_module_name)

Error compiling Cython file:
------------------------------------------------------------
...
#
# Copyright 2012-2019 CNRS-UM LIRMM, CNRS-AIST JRL
#

cimport eigen.eigen as eigen
       ^
------------------------------------------------------------

sva/sva.pxd:5:8: 'eigen/eigen.pxd' not found

Error compiling Cython file:
------------------------------------------------------------
...
#
# Copyright 2012-2019 CNRS-UM LIRMM, CNRS-AIST JRL
#

cimport eigen.c_eigen as c_eigen
       ^
------------------------------------------------------------

sva/c_sva.pxd:5:8: 'eigen/c_eigen.pxd' not found

Error compiling Cython file:
------------------------------------------------------------
...

cdef extern from "<SpaceVecAlg/SpaceVecAlg>" namespace "sva":
  cdef cppclass AdmittanceVec[T]:
    AdmittanceVec()
    AdmittanceVec(const AdmittanceVec[T] &)
    AdmittanceVec(const c_eigen.Matrix[T,c_eigen.six,c_eigen.one]&)
                       ^
------------------------------------------------------------

sva/c_sva.pxd:14:24: 'Matrix' is not a type identifier

[... 72 lines cut ...]
...
    AdmittanceVec(const c_eigen.Matrix[T,c_eigen.six,c_eigen.one]&)
    AdmittanceVec(const c_eigen.Matrix[T,c_eigen.three,c_eigen.one]&,
                  const c_eigen.Matrix[T,c_eigen.three,c_eigen.one]&)

    c_eigen.Matrix[T,c_eigen.three,c_eigen.one] angular() const
    c_eigen.Matrix[T,c_eigen.three,c_eigen.one] linear() const
                  ^
------------------------------------------------------------

sva/c_sva.pxd:19:19: invalid array declaration

Error compiling Cython file:
------------------------------------------------------------
...
    AdmittanceVec(const c_eigen.Matrix[T,c_eigen.three,c_eigen.one]&,
                  const c_eigen.Matrix[T,c_eigen.three,c_eigen.one]&)

    c_eigen.Matrix[T,c_eigen.three,c_eigen.one] angular() const
    c_eigen.Matrix[T,c_eigen.three,c_eigen.one] linear() const
    c_eigen.Matrix[T,c_eigen.six,c_eigen.one] vector() const
   ^
------------------------------------------------------------

```

gergondet, 2022-09-07:
Hello Yuquan,

Can you attach the full log? The issue seems to come from the installation of eigen3topython.

You can also enforce the selection of Python 3 which should fix the issue.

wyqsnddd, 2022-09-07:
Hi Pierre, 

Thanks for the fast reply. Indeed, I solved the issue by uncomment the following: `export PYTHON_FORCE_PYTHON3="true"`. Now the installation is moving on nicely. The log of the previous issue is here:

[build_and_install_warnings-2022-09-07-14-18-32.log](https://github.com/jrl-umi3218/mc_rtc/files/9506087/build_and_install_warnings-2022-09-07-14-18-32.log)

closed 2022-09-07 by wyqsnddd