jrl-umi3218/mc_rtc #331: Error in upgrading the apt package of libmc-rtc

mmurooka, 2023-01-18:
```bash
$ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  libmc-rtc1 linux-hwe-5.4-headers-5.4.0-132
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libmc-rtc
The following NEW packages will be installed:
  libmc-rtc
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
12 not fully installed or removed.
Need to get 0 B/2187 kB of archives.
After this operation, 13.3 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 306194 files and directories currently installed.)
Preparing to unpack .../libmc-rtc_2.0.0-1debian1-20230117125629-1_amd64.deb ...
Unpacking libmc-rtc (2.0.0-1debian1-20230117125629-1) ...
dpkg: error processing archive /var/cache/apt/archives/libmc-rtc_2.0.0-1debian1-20230117125629-1_amd64.deb (--unpack):
 trying to overwrite '/usr/etc/mc_rtc.yaml', which is also in package libmc-rtc1 1.14.2-1debian1-20230112080642-1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libmc-rtc_2.0.0-1debian1-20230117125629-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

gergondet, 2023-01-18:
Can you try to remove libmc-rtc1 manually?

mmurooka, 2023-01-20:
> Can you try to remove libmc-rtc1 manually?

It cannot be removed
```bash
$ sudo apt remove  libmc-rtc1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libmc-rtc-dev : Depends: libmc-rtc (= 2.0.0-1debian1-20230117125629-1) but it is not going to be installed
 mc-rtc-gui : Depends: libmc-rtc but it is not going to be installed
 mc-rtc-utils : Depends: libmc-rtc (= 2.0.0-1debian1-20230117125629-1) but it is not going to be installed
 python-mc-rtc : Depends: libmc-rtc but it is not going to be installed
 python3-mc-rtc : Depends: libmc-rtc but it is not going to be installed
 ros-melodic-mc-convex-visualization : Depends: libmc-rtc but it is not going to be installed
 ros-melodic-mc-log-visualization : Depends: libmc-rtc but it is not going to be installed
 ros-melodic-mc-rtc-rviz-panel : Depends: libmc-rtc but it is not going to be installed
 ros-melodic-mc-rtc-ticker : Depends: libmc-rtc but it is not going to be installed
 ros-melodic-mc-surfaces-visualization : Depends: libmc-rtc but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

gergondet, 2023-01-20:
The commit above should fix it. I'm reopening this until it is confirmed to work

mmurooka, 2023-01-20:
Thank you, apt upgrade succeeded.

commit 6c294e71 (2023-01-20): [debian] Add replaces field for libmc-rtc1

This should fix #331
closed 2023-01-20 by mmurooka