jrl-umi3218/mc_rtc #333: [tests] TestPython3Controller fails

mmurooka, 2023-01-23:
After updating everything with mc-rtc-superbuild, I got the following error.
```bash
$ ctest -R TestPython3Controller -V
UpdateCTestConfiguration  from :/home/mmurooka/workspace/build/mc_rtc/DartConfiguration.tcl
Parse Config file:/home/mmurooka/workspace/build/mc_rtc/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/mmurooka/workspace/build/mc_rtc/DartConfiguration.tcl
Parse Config file:/home/mmurooka/workspace/build/mc_rtc/DartConfiguration.tcl
Test project /home/mmurooka/workspace/build/mc_rtc
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added cleanup test TestPython3Log required by fixture TestPython3Log
Added 1 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 77
    Start 77: TestPython3Controller

77: Test command: /usr/bin/cmake "-E" "env" "PYTHONPATH=/home/mmurooka/workspace/build/mc_rtc/binding/python/mc_rtc/python3/RelWithDebInfo:/home/mmurooka/workspace/src/mc_rtc/tests/controllers:/home/mmurooka/workspace/src/catkin_data_ws/devel/lib/python3/dist-packages:/opt/ros/noetic/lib/python3/dist-packages:/home/mmurooka/workspace/install/lib/python3.8/site-packages:/home/mmurooka/workspace/install/lib/python3.8/dist-packages" "/usr/bin/cmake" "-E" "env" "LD_LIBRARY_PATH=/home/mmurooka/workspace/build/mc_rtc/src/mc_control/fsm:/home/mmurooka/workspace/build/mc_rtc/src:/home/mmurooka/workspace/src/catkin_ws/devel/lib:/home/mmurooka/workspace/src/catkin_data_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu:/home/mmurooka/workspace/install/lib:/usr/local/cuda/lib64" "/home/mmurooka/workspace/build/mc_rtc/tests/test_controller_ticker" "--run_test=RUN" "--" "/home/mmurooka/workspace/build/mc_rtc/tests/controllers/TestPython3/mc_rtc-TestPython3.conf" "400" ""
77: Working Directory: /home/mmurooka/workspace/build/mc_rtc/tests/controllers
77: Test timeout computed to be: 1500
77: Running 1 test case...
77: [info] Loading default global configuration /home/mmurooka/workspace/install/etc/mc_rtc.yaml
77: [info] Loading additional global configuration /home/mmurooka/.config/mc_rtc/mc_rtc.yaml
77: [info] Loading additional global configuration /home/mmurooka/workspace/build/mc_rtc/tests/controllers/TestPython3/mc_rtc-TestPython3.conf
77: [warning] Aliases env/beam was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/box was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/box_table was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/chair was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/corniche was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/crapahut was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/crapahut_bricks was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/door was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/drc_terrain was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/ground was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/litter was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/platform was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/pole was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/printer was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/seesaw was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/table was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/table_printer was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/valve was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases env/valve_handle was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/env_aliases.yml will prevail
77: [warning] Aliases object/box was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/door was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/litter was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/pole was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/printer was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/seesaw was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/valve was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [warning] Aliases object/valve_handle was already declared, new declaration from /home/mmurooka/workspace/build/mc_rtc/src/mc_robots/aliases/object_aliases.yml will prevail
77: [info] Enabled plugins: Plugin0
77: [info] GUI server enabled
77: [info] Will serve data on:
77: [info] - ipc:///tmp/mc_rtc_pub.ipc
77: [info] - tcp://*:4242
77: [info] Will handle requests on:
77: [info] - ipc:///tmp/mc_rtc_rep.ipc
77: [info] - tcp://*:4343
77: [info] Loading additional plugin configuration: /home/mmurooka/workspace/build/mc_rtc/tests/controllers/TestPlugin/etc/Plugin0.yaml
77: [info] Create controller Python3
77: [info] [PythonController] Running with Python 3.8.10
77: [info] [PythonController] sys.path: ['/home/mmurooka/workspace/build/mc_rtc/binding/python/mc_rtc/python3/RelWithDebInfo', '/home/mmurooka/workspace/src/mc_rtc/tests/controllers', '/home/mmurooka/workspace/src/catkin_data_ws/devel/lib/python3/dist-packages', '/opt/ros/noetic/lib/python3/dist-packages', '/home/mmurooka/workspace/install/lib/python3.8/site-packages', '/home/mmurooka/workspace/install/lib/python3.8/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/mmurooka/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
77: [critical] boundedSpeed is already handled by another loading function
77: [critical] === Backtrace ===
77:  0# void mc_rtc::log::error_and_throw<std::runtime_error, char const (&) [50], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(char const (&) [50], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/mmurooka/workspace/src/mc_rtc/include/mc_rtc/logging.h:52
[... cut ...]
Failed test dependencies: TestPython3Controller
2/2 Test #78: TestPython3Log ...................***Not Run   0.00 sec

0% tests passed, 2 tests failed out of 2

Total Test time (real) =   0.60 sec

The following tests FAILED:
	 77 - TestPython3Controller (Failed)
	 78 - TestPython3Log (Not Run)
Errors while running CTest
Output from these tests are in: /home/mmurooka/workspace/build/mc_rtc/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

gergondet, 2023-01-23:
Hi @mmurooka 

This could be due to one of two things:
- ABI issue with the Python bindings, that is because incremental builds of the bindings are not super reliable now, can you try to run `ninja force-mc_rtc-python3-bindings` in mc_rtc build folder?
- another issue that could explain why you get `unknown location(0): fatal error: in "RUN": std::runtime_error: boundedSpeed is already handled by another loading function` is that you have components loading old versions of mc_rtc at runtime along the current version (e.g some plugins still linked to 1.x)

The later is a little unlikely these days because mc_rtc install rules systematically removes the previous installation (to avoid this specific issue among others).

I'm working towards a PR to improve incremental builds of Python bindings in https://github.com/gergondet/mc_rtc/tree/topic/ImproveCythonIncrementalBuilds which should hopefully circumvent the first issue for good (there's a long standing hack in the build_and_install script to trigger the force binding target on test failures)

mmurooka, 2023-01-23:
Thanks for the explanation.

> ABI issue with the Python bindings, that is because incremental builds of the bindings are not super reliable now, can you try to run ninja force-mc_rtc-python3-bindings in mc_rtc build folder?

After doing this, the tests passed.

closed 2023-01-23 by mmurooka