jrl-umi3218/mc_rtc #337: Questions about frame

mmurooka, 2023-01-25:
This is not a issue, it is a question. I haven't been able to catch up much since the concept `frame` was introduced in mc_rtc, so I have a few questions.

- Should the following two still return the same pose?
https://github.com/jrl-umi3218/mc_rtc/blob/c45e8e4d6164f45e9eb1a747ef5084e49dc9abcd/src/mc_rbdyn/Robot.cpp#L1026-L1029
https://github.com/jrl-umi3218/mc_rtc/blob/c45e8e4d6164f45e9eb1a747ef5084e49dc9abcd/include/mc_tasks/TransformTask.h#L104-L108

- Where in the code is the `frame_->position` updated each control cycle?

gergondet, 2023-01-25:
> Should the following two still return the same pose?

If the controlled frame is a surface then yes.

> Where in the code is the frame_->position updated each control cycle?

There is no update mechanism. It is computed on demand [here](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_rbdyn/RobotFrame.cpp#L37)

mmurooka, 2023-01-25:
Oh, I see. Thank you for your explanation.

closed 2023-01-25 by mmurooka