jrl-umi3218/mc_rtc #338: MCControl.so cannot find libmc_rbdyn.so.2

TsuruMasato, 2023-02-02:
Hello, this is Tsuru.

In Choreonoid simulation, I am facing the avobe error.

I updated mc_rtc to the latest today. It passed all tests smoothly.
However, when I run my mc controller with Choreonoid, it always crashes at the beginning.

The error log says : 
 "`mc_control.so` cannot open a shared object file `libmc_rbdyn.so.1`.  No such files or directory."
(Sorry , the picture contains some Japanese...)

[image]

Does it mean the mc_control.so is still linked to an old `libmc_rbdyn.so.1`...?

There is a new .so, `libmc_rbdyn.so.2` at the correct place, as far as I see.

[image]

At least, `libmc_rbdyn.so` and `libmc_control.so` seem to be connected to the newer one...
[image]

So..., could anyone tell me how to solve this old linking problem...?
Sould I just delete the old .so file and re-build the whole mc_rtc ?

Sorry for disturbing.
Thank you in advance.

gergondet, 2023-02-02:
Did you rebuild mc_openrtm?

TsuruMasato, 2023-02-02:
Ahhhh, I was extremely stupid...
It worked...
Thank you very much...

closed 2023-02-02 by TsuruMasato