jrl-umi3218/mc_rtc #342: Segmentation fault on loading robot configuration

mmurooka, 2023-02-08:
The mc_rtc controller sometimes has a segmentation fault at startup, although I have not yet found a clear reproduction condition. The backtrace printed by gdb is as follows. Strangely, this happened when I added surfaces in the robot rsdf file, or when I added the links in the environment URDF model. This problem occurred with [MultiContactController](https://github.com/isri-aist/MultiContactController), but I do not think it is controller-dependent.

```bash
Reading symbols from ./mc_rtc_ticker...
[New LWP 2514211]
[New LWP 2514217]
[New LWP 2514216]
[New LWP 2514214]
[New LWP 2514215]
[New LWP 2514213]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./mc_rtc_ticker'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::GetStringLength (this=0x0)
    at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1991
1991	    bool StringEqual(const GenericValue<Encoding, SourceAllocator>& rhs) const {
[Current thread is 1 (Thread 0x7efd4d874240 (LWP 2514211))]
(gdb) bt
#0  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::GetStringLength() const (this=0x0)
    at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1991
#1  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::StringEqual<rapidjson::CrtAllocator>(rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator> const&) const (rhs=..., this=0x7ffec3c486b0) at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1996
#2  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::FindMember<rapidjson::CrtAllocator>(rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator> const&) (name=..., this=<optimized out>) at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1158
#3  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::FindMember<rapidjson::CrtAllocator>(rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator> const&) const (name=..., this=<optimized out>) at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1162
#4  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::FindMember(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const (name=<error reading variable: Cannot access memory at address 0x1d>, this=<optimized out>)
    at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1173
#5  rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::CrtAllocator>::HasMember(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const (name=<error reading variable: Cannot access memory at address 0x1d>, this=<optimized out>)
    at /home/mmurooka/workspace/src/mc_rtc/3rd-party/RapidJSON/rapidjson/document.h:1105
#6  mc_rtc::Configuration::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const
    (this=this@entry=0x556c8db3dc20, key="init_pos") at /home/mmurooka/workspace/src/mc_rtc/src/mc_rtc/Configuration.cpp:205
#7  0x00007efd51ebd436 in mc_control::MCController::<lambda(const string&, const mc_rtc::Configuration&)>::operator()(const std::string &, const mc_rtc::Configuration &) (robotName="jvrc1", config=..., __closure=<optimized out>) at /usr/include/c++/9/bits/basic_string.h:267
#8  0x00007efd51ed0474 in mc_control::MCController::MCController(std::vector<std::shared_ptr<mc_rbdyn::RobotModule>, std::allocator<std::shared_ptr<mc_rbdyn::RobotModule> > > const&, double, mc_rtc::Configuration const&, mc_control::ControllerParameters)
    (this=0x556c8d37a740, robots_modules=..., dt=<optimized out>, config=..., params=...) at /home/mmurooka/workspace/src/mc_rtc/src/mc_control/MCController.cpp:253
#9  0x00007efd45edff0d in mc_control::fsm::Controller::Controller(std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration const&, mc_control::ControllerParameters) (this=0x556c8d37a740, rm=..., dt=0.0050000000000000001, config=..., params=...) at /usr/include/c++/9/bits/stl_uninitialized.h:82
#10 0x00007efd45e05b19 in MCC::MultiContactController::MultiContactController(std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration const&)
    (this=0x556c8d37a740, rm=..., dt=0.0050000000000000001, _config=...) at /usr/include/c++/9/ext/atomicity.h:96
#11 0x00007efd45ea6800 in create(std::string const&, mc_rbdyn::RobotModulePtr const&, double const&, mc_control::Configuration const&)
    (robot=..., dt=@0x7ffec3c49158: 0.0050000000000000001, conf=...) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:170
#12 0x00007efd51f02ae2 in mc_rtc::ObjectLoader<mc_control::MCController>::create_from_handles<std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration)
    (this=this@entry=0x556c8d348810, name="MultiContactController") at /usr/include/c++/9/string_view:129
#13 0x00007efd51f03885 in mc_rtc::ObjectLoader<mc_control::MCController>::create<std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration)
    (this=this@entry=0x556c8d348810, name="MultiContactController") at /home/mmurooka/workspace/src/mc_rtc/include/mc_rtc/Configuration.h:116
#14 0x00007efd51ef6c85 in mc_rtc::ObjectLoader<mc_control::MCController>::create_object<std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>, double, mc_rtc::Configuration)
    (name="MultiContactController", this=0x556c8d348810) at /usr/include/c++/9/bits/shared_ptr_base.h:756
#15 mc_control::MCGlobalController::AddController(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=0x7ffec3c49c90, name="MultiContactController") at /home/mmurooka/workspace/src/mc_rtc/src/mc_control/mc_global_controller.cpp:990
#16 0x00007efd51ef7f56 in mc_control::MCGlobalController::MCGlobalController(mc_control::MCGlobalController::GlobalConfiguration const&)
    (this=0x7ffec3c49c90, conf=...) at /home/mmurooka/workspace/src/mc_rtc/src/mc_control/mc_global_controller.cpp:86
#17 0x00007efd51ef8613 in mc_control::MCGlobalController::MCGlobalController(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) (this=0x7ffec3c49c90, conf=..., rm=...) at /usr/include/c++/9/ext/atomicity.h:96
--Type <RET> for more, q to quit, c to continue without paging--
#18 0x0000556c8b82fe2c in main() () at /usr/include/c++/9/bits/shared_ptr_base.h:616
```

The terminal message at controller startup is as follows. Note that robot-specific configuration files are not placed in this controller.
```bash
$ rosrun mc_rtc_ticker mc_rtc_ticker
[info] Loading default global configuration /home/mmurooka/workspace/install/etc/mc_rtc.yaml
[info] Loading additional global configuration /home/mmurooka/.config/mc_rtc/mc_rtc.yaml
[info] Enabled plugins: ROS (autoload)
[info] GUI server enabled
[info] Will serve data on:
[info] - ipc:///tmp/mc_rtc_pub.ipc
[info] - tcp://*:4242
[info] Will handle requests on:
[info] - ipc:///tmp/mc_rtc_rep.ipc
[info] - tcp://*:4343
[info] Loading additional plugin configuration: /home/mmurooka/workspace/install/lib/mc_plugins/etc/ROS.yaml
[info] Loading additional controller configuration: /home/mmurooka/ros/ws_bwc/devel/.private/multi_contact_controller/lib/mc_controller/etc/MultiContactController.yaml
[error] LIPMWalking was compiled with 2.0.1 but mc_rtc is currently at version 2.1.0, you might experience subtle issues and should recompile your code
[info] Create controller MultiContactController
[success] [overwriteConfig] Overwrite with configuration jvrc1.
Segmentation fault
```

There seems to be a problem accessing configuration in the following line.

https://github.com/jrl-umi3218/mc_rtc/blob/b24fc0a340a77a867a05ae3d5857592846cb95b0/src/mc_control/MCController.cpp#L253

The problem appears to have been avoided by using the following patch to prevent the addition of empty robot configurations. However, my understanding is that there should be no problem if empty configurations are added, so I don't know why the problem occurs.

https://github.com/jrl-umi3218/mc_rtc/compare/master...mmurooka:robot-config-debug

gergondet, 2023-02-08:
Thanks for reporting the issue. It was a subtle problem with robot's configuration loading as explained in https://github.com/jrl-umi3218/mc_rtc/commit/a96d61aaa223e672c33bbb2109541e0498a4d4bf description.

closed 2023-02-08 by gergondet

mmurooka, 2023-02-08:
Thanks for the fix @gergondet 
I have confirmed that the problem no longer occurs with or without the https://github.com/isri-aist/MultiContactController/commit/45838c6528b8b73105b0cd35e018b1676bb07bf4 change.