jrl-umi3218/mc_rtc #343: build_and_install.sh script fails

rohanpsingh, 2023-02-08:
I'm trying to update mc-rtc using the `build_and_install.sh` script. It seems to fail a this step:
```
$ /home/rohan/src/mc_rtc/build/tests/test_controller_ticker --run_test=CONSTRUCTION_FAILURE -- /home/rohan/src/mc_rtc/build/tests/robots/NoCreateRobot/mc_rtc-NoCreateRobot.conf
Running 1 test case...
[info] Loading default global configuration /home/rohan/openrtp/etc/mc_rtc.yaml
[info] Loading additional global configuration /home/rohan/.config/mc_rtc/mc_rtc.yaml
[info] Loading additional global configuration /home/rohan/src/mc_rtc/build/tests/robots/NoCreateRobot/mc_rtc-NoCreateRobot.conf
[critical] Failed to resolve create symbol in /home/rohan/src/mc_rtc/build/tests/robots/NoCreateRobot/NoCreateRobot.so
[critical] === Backtrace ===
 0# void mc_rtc::log::error_and_throw<mc_rtc::LoaderException, char const (&) [38], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(char const (&) [38], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rohan/src/mc_rtc/include/mc_rtc/logging.h:52
 1# mc_rbdyn::RobotModule* mc_rtc::ObjectLoader<mc_rbdyn::RobotModule>::create_from_handles<>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rohan/src/mc_rtc/include/mc_rtc/loader.hpp:167
 2# std::shared_ptr<mc_rbdyn::RobotModule> mc_rtc::ObjectLoader<mc_rbdyn::RobotModule>::create_object<>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rohan/src/mc_rtc/include/mc_rtc/loader.hpp:227
 3# std::shared_ptr<mc_rbdyn::RobotModule> mc_rbdyn::RobotLoader::get_robot_module_from_lib<>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rohan/src/mc_rtc/include/mc_rbdyn/RobotLoader.h:214
 4# std::shared_ptr<mc_rbdyn::RobotModule> mc_rbdyn::RobotLoader::get_robot_module<>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rohan/src/mc_rtc/include/mc_rbdyn/RobotLoader.h:121
 5# mc_control::MCGlobalController::GlobalConfiguration::GlobalConfiguration(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) at /home/rohan/src/mc_rtc/src/mc_control/mc_global_controller_configuration.cpp:91
 6# mc_control::MCGlobalController::MCGlobalController(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) at /home/rohan/src/mc_rtc/src/mc_control/mc_global_controller.cpp:38
 7# CONSTRUCTION_FAILURE::test_method() at /usr/include/c++/7/bits/shared_ptr.h:93
 8# CONSTRUCTION_FAILURE_invoker at /home/rohan/src/mc_rtc/tests/test_controller_ticker.cpp:17
 9# 0x00007F1BB765C2CE in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
10# 0x00007F1BB765B77D in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
11# 0x00007F1BB765B861 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
12# 0x00007F1BB765BFDD in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
13# 0x00007F1BB768A8D1 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
14# 0x00007F1BB7666C6B in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
15# 0x00007F1BB7666E51 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
16# 0x00007F1BB765FCC8 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
17# 0x00007F1BB768813F in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
18# 0x00007F1BB53C2C87 in /lib/x86_64-linux-gnu/libc.so.6
19# 0x0000561A3A77DF7A in /home/rohan/src/mc_rtc/build/tests/test_controller_ticker

[critical] Failed to create NoCreateRobot to use as a main robot
[critical] === Backtrace ===
 0# void mc_rtc::log::error_and_throw<std::runtime_error, char const (&) [43], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>(char const (&) [43], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) at /home/rohan/src/mc_rtc/include/mc_rtc/logging.h:52
 1# mc_control::MCGlobalController::GlobalConfiguration::GlobalConfiguration(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) at /home/rohan/src/mc_rtc/src/mc_control/mc_global_controller_configuration.cpp:95
 2# mc_control::MCGlobalController::MCGlobalController(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) at /home/rohan/src/mc_rtc/src/mc_control/mc_global_controller.cpp:38
 3# CONSTRUCTION_FAILURE::test_method() at /usr/include/c++/7/bits/shared_ptr.h:93
 4# CONSTRUCTION_FAILURE_invoker at /home/rohan/src/mc_rtc/tests/test_controller_ticker.cpp:17
 5# 0x00007F1BB765C2CE in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
 6# 0x00007F1BB765B77D in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
 7# 0x00007F1BB765B861 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
 8# 0x00007F1BB765BFDD in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
 9# 0x00007F1BB768A8D1 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
10# 0x00007F1BB7666C6B in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
11# 0x00007F1BB7666E51 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
12# 0x00007F1BB765FCC8 in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
13# 0x00007F1BB768813F in /usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so.1.65.1
14# 0x00007F1BB53C2C87 in /lib/x86_64-linux-gnu/libc.so.6
15# 0x0000561A3A77DF7A in /home/rohan/src/mc_rtc/build/tests/test_controller_ticker
```

rohanpsingh, 2023-02-08:
This is the log file:
[build_and_install_warnings-2023-02-08-10-33-17.log](https://github.com/jrl-umi3218/mc_rtc/files/10680838/build_and_install_warnings-2023-02-08-10-33-17.log)

gergondet, 2023-02-08:
Hi @rohanpsingh 

The logs you pasted in the initial message are ok. Those tests are checking failure cases. However, there is indeed tests that do not pass.

The critical issue is:
```
59: [critical] boundedSpeed is already handled by another loading function
```

Can you check that your system does not have remnants of a 1.x installation? This would explain the multiple loading definitions.

rohanpsingh, 2023-02-08:
Thanks for the quick response~ @gergondet 

It seems I needed to completely delete mc-rtc build directory and then execute `build_and_install.sh`. It succeeds now. But I'm afraid other users may face the same problem...

closed 2023-02-08 by rohanpsingh

mmurooka, 2023-02-08:
Pointer: The same error is reported in https://github.com/jrl-umi3218/mc_rtc/issues/333