jrl-umi3218/mc_rtc #345: Incorrect transformation of environment collision shapes

mmurooka, 2023-02-11:
In some cases, the transformation of the collision shape of the environment model does not seem to be reflected correctly; the problem is confirmed by the fact that the arrow of the collision pair between the robot link and the environment model link is not drawn correctly in the "Collisions" tab in Rviz. The following image is an example of a situation where this problem occurs. A collision pair is specified between the robot's left-hand link and the ground link, but the green arrow does not correctly point to the ground link.

[image]

I tried four different settings in the matrix with the two options below, three of which did not correctly reflect the transformation of the collision shape.

1. Primitive shape (box, etc.) or mesh shape.
2. The offset is specified in a (fixed) joint tag or in a visual/collision tag.

- [debug-collision-transform/mesh-joint-offset](https://github.com/jrl-umi3218/mc_rtc_data/compare/master...mmurooka:mc_rtc_data:debug-collision-transform/mesh-joint-offset): OK
- [debug-collision-transform/mesh-geometry-offset](https://github.com/jrl-umi3218/mc_rtc_data/compare/master...mmurooka:mc_rtc_data:debug-collision-transform/mesh-geometry-offset): NG
- [debug-collision-transform/box-joint-offset](https://github.com/jrl-umi3218/mc_rtc_data/compare/master...mmurooka:mc_rtc_data:debug-collision-transform/box-joint-offset): NG
- [debug-collision-transform/box-geometry-offset](https://github.com/jrl-umi3218/mc_rtc_data/compare/master...mmurooka:mc_rtc_data:debug-collision-transform/box-geometry-offset): NG

This problem was found when adding a collision pair of robot and environment in [MultiContactController](https://github.com/isri-aist/MultiContactController).

gergondet, 2023-02-13:
Thanks for raising the issue. PR #347 fixes all the scenarios you presented. Note that the issue only affect fixed objects in the Tasks backend because Tasks does not update the sch object transformation in that case since it shouldn't be required.

The core of the issue is fixed by [`4568e09` (#347)](https://github.com/jrl-umi3218/mc_rtc/pull/347/commits/4568e091157c7aeb9135951f9e5277ddc6cf86e9) which correctly initialize all objects transformation after the initialization.

The [`3117bd2` (#347)](https://github.com/jrl-umi3218/mc_rtc/pull/347/commits/3117bd20c5e49aa056dadfc67ae51f21967fd4d4) commit fixes the first `NG` issue. In that case the transform was ignored because only collision transform were considered. With this commit, the visual transformation will be considered if no collision is provided.

mmurooka, 2023-02-13:
Thanks, I have confirmed that the collision transformation of the MultiContactController sample is correct with the patch.

PR #347 (2023-02-13): [mc_rbdyn] Fix collision objects after fixing collision transforms
Closes #345
closed 2023-02-13 by gergondet