jrl-umi3218/mc_rtc #348: Additional 3D markers in the GUI

mmurooka, 2023-02-13:
This is not an issue, but a feature request for the future.

It would be nice to have more types of 3D markers in the GUI. For example, the following types supported by ROS [`visualization_msgs/Marker` messages](http://wiki.ros.org/rviz/DisplayTypes/Marker) are not available in mc_rtc.

- boxes (`CUBE` in `visualization_msgs/Marker`)
- spheres and ellipsoids (`SPHERE` in `visualization_msgs/Marker`)
- lines (`LINE_LIST` in `visualization_msgs/Marker`. Although `LINE_STRIP` can be handled by `mc_rtc::gui::Trajectory`, `LINE_LIST` is still useful.)

Supporting these features reduces the motivation for users to do things away from the mc_rtc concept, such as publishing their own ROS topics for the GUI as in https://github.com/isri-aist/Motion6DoF/pull/5#issuecomment-1418423577.

gergondet, 2023-02-13:
> boxes (CUBE in visualization_msgs/Marker)
> spheres and ellipsoids (SPHERE in visualization_msgs/Marker)

These two can (mostly) be handled via [`Visual`](https://github.com/jrl-umi3218/mc_rtc/blob/master/include/mc_rtc/gui/Visual.h) although:
- this is not part of this [documentation](https://jrl-umi3218.github.io/mc_rtc/tutorials/usage/gui.html)
- creating Visual objects is a little more involved than they should be
- we cannot display ellipsoid via `Sphere` (since the `Visual` element is from the definition of URDF visual elements) but we can via `Superellipsoid` although I think the client does not handle them

Note: mc_rtc's `Visual` also handles cylinders and meshes

I think we can add a nice helper to simplify usage here (and improve the docs)

To handle `LINE_LIST` we have to introduce a new type in mc_rtc GUI.

mmurooka, 2023-03-01:
I will close this as it was handled by https://github.com/jrl-umi3218/mc_rtc/pull/299. Thank you.

gergondet, 2023-03-02:
@mmurooka how about `LINE_LIST`-like functionality?

mmurooka, 2023-03-02:
Oh, yes. It is desirable to have `LINE_LIST` as well. (At least Motion6DoF also used `LINE_LIST`.) I'll reopen the issue.