jrl-umi3218/mc_rtc #350: my_first_controller.py failed

zzz622848, 2023-03-01:
PYTHONPATH=$HOME/my_python_controllers:$PYTHONPATH mc_mujoco
...
[warning] Controller Python#my_first_controller.MyFirstController enabled in configuration but not available
...
[error] No controller selected or selected controller is not enabled, please check your configuration file
[info] Note common reasons for this error include:
	- The controller name does not match the name exported by CONTROLLER_CONSTRUCTOR
	- The controller library is not in a path read by mc_rtc
	- The controller constuctor segfaults
	- The controller library hasn't been properly linked
[critical] No controller enabled

mc_rtc.yaml:
MainRobot: JVRC1
Enabled: [Python#my_first_controller.MyFirstController]

gergondet, 2023-03-01:
Hi @zzz622848 

If you are enforcing Python3 can you try `Python3#my_first_controller.MyFirstController` instead?

zzz622848, 2023-03-01:
fixed

zzz622848, 2023-03-01:
fixed

closed 2023-03-01 by zzz622848