jrl-umi3218/mc_rtc #352: python: callback of mc_rtc.gui.Transform is empty

zzz622848, 2023-03-02:
self.mypoint3d_ = [0., 0., 0., 0., 1., 1., 1.]
self.gui().addElement(["Python"], mc_rtc.gui.Transform("Point3D", self.mypoint3d, self.mypoint3d))

  def mypoint3d(self, s=None):
    if s is None:
        return self.mypoint3d_
    else:
        print("Change mypoint3d to {}".format(s))
        # self.mypoint3d_ = s

s value is empty([])  when callback function mypoint3d

gergondet, 2023-03-02:
Thanks for raising the issue. It should be fixed in the commit referenced above

Note that the `s` value you receive will be an `sva.PTransformd` and you'll have to form a 7d array from that if you want to keep that representation:
```
q = eigen.Quaterniond(s.rotation())
t = s.translation()
self.mypoint3d_ = [q.w(), q.x(), q.y(), q.z(), t.x(), t.y(), t.z()]
```

closed 2023-03-02 by gergondet

zzz622848, 2023-03-03:
Sorry to bother you again.
I have tried your advise, but zhe feedback s value is still empty.Does something wrong with my code?
`
self.mypoint3d_ = [1., 0., 0., 0, 2, 2, 2]
self.gui().addElement(["Python"], mc_rtc.gui.Transform("Point3D", self.mypoint3d, self.mypoint3d))

def mypoint3d(self, s=None):
    if s is None:
        return self.mypoint3d_
    else:
        q = eigen.Quaterniond(s.rotation())
        t = s.translation()
        self.mypoint3d_ = [q.w(), q.x(), q.y(), q.z(), t.x(), t.y(), t.z()]
`
Output is:
q = eigen.Quaterniond(s.rotation())
AttributeError: 'list' object has no attribute 'rotation'

s value is still empty([])

zzz622848, 2023-03-03:
By the way, is there any full tutorials about python version of mc_rtc, or some code i can refer to that?

zzz622848, 2023-03-03:
I have tested mc_rtc.gui.Point3D and mc_rtc.gui.Rotation, they will give the callback s value [3] and [9] respectively.But mc_rtc.gui.Transform will give s value [0].

gergondet, 2023-03-03:
Did you update mc_rtc? You need to have the change in https://github.com/jrl-umi3218/mc_rtc/commit/aa5cb798a9c7f6668f2a221d03789b853c0b5e2a for this to work correctly

zzz622848, 2023-03-04:
> Did you update mc_rtc? You need to have the change in [aa5cb79](https://github.com/jrl-umi3218/mc_rtc/commit/aa5cb798a9c7f6668f2a221d03789b853c0b5e2a) for this to work correctly

Thanks very much. Fixed this problem after update new code.

commit aa5cb798 (2023-03-02): [bindings] Avoid Transform callback receiving empty values

Closes #352