jrl-umi3218/mc_rtc #358: Is there any way to change dimWeight of CoM task from .yaml file + mc_rtc::Configruation?

TsuruMasato, 2023-03-28:
This is Tsuru.

Z direction of CoM task is less important than x & y directions.
So that, I am changing the task weight of Z direction as following in my source code:
`comTask_->dimWeight(Eigen::Vector3d(1.0, 1.0, 0.5));`

I wonder if dimWeight is adjustable from yaml file...

Thank you.

gergondet, 2023-03-28:
Hi @TsuruMasato 

Yes, you can specify `dimWeight` in the yaml (see the yaml reference here: https://jrl-umi3218.github.io/mc_rtc/json.html#MetaTask/CoMTask)

```yaml
CoMTask:
  type: com
  dimWeight: [1.0, 1.0, 0.5]
```

TsuruMasato, 2023-03-28:
Sorry, I was too stupid.
I couldn't find that document page when I searched for...
Thank you so much.

closed 2023-03-28 by TsuruMasato