jrl-umi3218/mc_rtc #361: [critical] firstOrderImpedance is already handled by another loading function

853734690, 2023-04-20:
Hello, Dr.Gergondet @gergondet  and Dr.Tanguy @arntanguy，I am so sorry to trouble you.
   My current research topic is that use [BaselineWalkingController](https://github.com/isri-aist/BaselineWalkingController) for my robot .everything went smoothly yesterday . But today when  I click the start simulation botton in the choreonoid  , I get the following error in the terminal . It is shown the firstOrderImpedance is already handled by another loading function.
  And then I go back to everything I've done before ,  I realized that I made the mistake of adding the firstOrderImpedance task by mc_rtc GUI ,but actually the tasks is already loaded  by [BaselineWalkingController](https://github.com/isri-aist/BaselineWalkingController) .I think this is the main reason.

`
[critical] firstOrderImpedance is already handled by another loading function
[critical] === Backtrace ===
 0# void mc_rtc::log::error_and_throw<std::runtime_error, char const (&) [50], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(char const (&) [50], std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /usr/include/mc_rtc/logging.h:52
 1# mc_solver::GenericLoader<mc_tasks::MetaTaskLoader, mc_tasks::MetaTask>::register_load_function(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::shared_ptr<mc_tasks::MetaTask> (mc_solver::QPSolver&, mc_rtc::Configuration const&)>) at /usr/include/mc_solver/GenericLoader.hpp:124
 2# __static_initialization_and_destruction_0 at /home/zxa/ros/ws_bwc/src/isri-aist/BaselineWalkingController/src/tasks/FirstOrderImpedanceTask.cpp:130
 3# 0x00007F3ED5C648D3 in /lib64/ld-linux-x86-64.so.2
 4# 0x00007F3ED5C6939F in /lib64/ld-linux-x86-64.so.2
 5# 0x00007F3ED4DD316F in /lib/x86_64-linux-gnu/libc.so.6
 6# 0x00007F3ED5C6896A in /lib64/ld-linux-x86-64.so.2
 7# 0x00007F3ED0A4DF96 in /lib/x86_64-linux-gnu/libdl.so.2
 8# 0x00007F3ED4DD316F in /lib/x86_64-linux-gnu/libc.so.6
 9# 0x00007F3ED4DD31FF in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007F3ED0A4E745 in /lib/x86_64-linux-gnu/libdl.so.2
11# 0x00007F3ED0A4E051 in /lib/x86_64-linux-gnu/libdl.so.2
12# 0x00007F3EBDF5214A in /usr/lib/x86_64-linux-gnu/libltdl.so.7
13# 0x00007F3EBDF4ECA2 in /usr/lib/x86_64-linux-gnu/libltdl.so.7
14# 0x00007F3EBDF50984 in /usr/lib/x86_64-linux-gnu/libltdl.so.7
15# 0x00007F3EBDF51121 in /usr/lib/x86_64-linux-gnu/libltdl.so.7
16# 0x00007F3E677993F7 in /usr/lib/x86_64-linux-gnu/libmc_rtc_loader.so.2
17# 0x00007F3E677B0E75 in /usr/lib/x86_64-linux-gnu/libmc_rtc_loader.so.2
18# 0x00007F3E67799591 in /usr/lib/x86_64-linux-gnu/libmc_rtc_loader.so.2
19# 0x00007F3E67799E3E in /usr/lib/x86_64-linux-gnu/libmc_rtc_loader.so.2
20# 0x00007F3E716909E2 in /usr/lib/x86_64-linux-gnu/libmc_control.so.2
21# 0x00007F3E7168B60A in /usr/lib/x86_64-linux-gnu/libmc_control.so.2
22# 0x00007F3E7168C00D in /usr/lib/x86_64-linux-gnu/libmc_control.so.2
23# 0x00007F3E71923273 in /usr/lib/MCControl.so
24# 0x00007F3E71946901 in /usr/lib/MCControl.so
25# 0x00007F3EA98F362A in /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
26# 0x00007F3EA9905BB0 in /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
27# 0x00007F3EA992A4AF in /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
28# 0x00007F3EA9A0D32C in /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
29# 0x00007F3EAAB03E64 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
30# 0x00007F3EA9A11B08 in /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
31# 0x00007F3EAAAFC9E2 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
32# 0x00007F3EAAAD6026 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
33# 0x00007F3EAAB24390 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
34# 0x00007F3EAAB25398 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
35# 0x00007F3EAAB212EA in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
36# 0x00007F3EAAAC8350 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
37# 0x00007F3EAAAC951F in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
38# 0x00007F3EAAAC7EC9 in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
39# 0x00007F3EAAAC949E in /usr/lib/x86_64-linux-gnu/libomniORB4.so.2
40# 0x00007F3EAA80B4C8 in /usr/lib/x86_64-linux-gnu/libomnithread.so.4
41# 0x00007F3ED3A1C6DB in /lib/x86_64-linux-gnu/libpthread.so.0
42# 0x00007F3ED4D8D61F in /lib/x86_64-linux-gnu/libc.so.6

omniORB: (1096) 2023-04-20 13:21:22.502629: Error: a servant has been deleted that is still activated.
      id: root<59> (active)
`
  And then I rebuild my controller but It doesn't solve the problem, and my choreonoid project cannot  load MCcontroller components as before.and then I select a  sample controllers without the firstOrderImpedance task ,and get another error in the terminal:  
`choreonoid: symbol lookup error: /usr/lib/x86_64-linux-gnu/libTasks.so.1: undefined symbol: ql_`

Could you give me some advice?

gergondet, 2023-04-20:
Hi @853734690 ,

You should update `BaselineWalkingController`, the `FirstOrderImpedanceTask` has been added to `mc_rtc` and this is complaining about that (the task loader is being registered once in mc_rtc and once in the controller which is not allowed)

> symbol lookup error: /usr/lib/x86_64-linux-gnu/libTasks.so.1: undefined symbol: ql_

This is strange. The `ql_` symbol was removed from `eigen-qld` a while back and the Tasks library should have been rebuilt without the symbol. How did you install and update mc_rtc?

gergondet, 2023-04-20:
Could you check the output of `dpkg-query -l libtasks-qld-dev libeigen-qld-dev` and also did you install mc_rtc packages manually on the side?

853734690, 2023-04-20:
I don't have any record of install mc_rtc packages manually . I check the out put and found that  Tasks library maybe have not  been rebuilt successfully. 

```
==================================================================================================
ii  libeigen-qld-dev                               1.2.3-1debian1-2023031309052 amd64                        eigen-qld - development files
ii  libtasks-qld-dev                               1.7.2-1debian1-2023012408242 amd64                        tasks - development f
==================================================================================================
```

And today I update `BaselineWalkingController` , there was an error show in the terminal:

```
Errors     << baseline_walking_controller:make /home/zxa/ros/ws_bwc/logs/baseline_walking_controller/build.make.000.log
/usr/bin/ld: warning: libyaml-cpp.so.0.5, needed by /usr/lib/x86_64-linux-gnu/libmc_rtc_utils.so.2.4.0, may conflict with libyaml-cpp.so.0.7
/usr/lib/x86_64-linux-gnu/libTVM.so.0.9.0：undefined reference to ‘ql_’
collect2: error: ld returned 1 exit status
```
 maybe they are the same reason ? I'm only new to this knowledge，hope to  get your reply .

Thanks in advance

gergondet, 2023-04-20:
Hi @853734690 

Yes. The tasks package is outdated (it was generated before the update to your eigen-qld package). Can you run `apt upgrade` and make sure it gets updated?  The same issue is probably affecting your TVM package (you can check the package date in `dpkg-query -l libtvm-dev`

853734690, 2023-04-21:
Thank You for your reply.
I have try it and everything goes well again!
Thank You
Best Regards

closed 2023-04-21 by 853734690