jrl-umi3218/mc_rtc #387: Request for Assistance: Adding a New Controller to mc_rtc

Saeed-Mansouri, 2023-09-02:
I am seeking assistance with adding a new controller to `mc_rtc`. Here are the steps I've taken so far:

Copied the `lipm_walking_controller` folder and pasted it.
Renamed the copied folder to `xo_controller`.
Changed the controller's defined name from "LIPMWalking" to "XO".
Removed the build folders for both `mc_rtc` and `xo_controller`.
Rebuilt the projects.
However, I encountered the following errors when attempting to run the new controller:

```
rosrun mc_rtc_ticker mc_rtc_ticker
[info] Loading default global configuration /usr/local/etc/mc_rtc.yaml
[info] Loading additional global configuration /home/hmr/.config/mc_rtc/mc_rtc.conf
[info] GUI server enabled
[info] Will serve data on:
[info] - ipc:///tmp/mc_rtc_pub.ipc
[info] - tcp://*:4242
[info] Will handle requests on:
[info] - ipc:///tmp/mc_rtc_rep.ipc
[info] - tcp://*:4343
[info] Enabled plugins: ROS (autoload)
[info] Loading additional plugin configuration: /usr/local/lib/mc_plugins/etc/ROS.yaml
[info] Loading additional controller configuration: /usr/local/lib/mc_controller/etc/XO.yaml
[warning] Controller XO enabled in configuration but not available
[info] Loading additional plugin configuration: /usr/local/lib/mc_plugins/etc/footsteps_planner_plugin.yaml
[info] Loading additional plugin configuration: /usr/local/lib/mc_plugins/etc/ExternalFootstepPlannerPlugin.yaml
[info] Create controller HalfSitPose
[info] Robots loaded in controller:
[info] - XORobot
[info] - ground
[info] MCController(base) ready
[info] Added task posture_XORobot
[warning] [MC_RTC_DEPRECATED] Prefer using addContact/removeContact for contacts manipulation instead of QPSolver::setContacts
[success] MCHalfSitPoseController init done
[warning] [MCController::HalfSitPose] No state observation pipeline configured: the state of the real robots will not be estimated
[error] No controller selected or selected controller is not enabled, please check your configuration file
[info] Note common reasons for this error include:
	- The controller name does not match the name exported by CONTROLLER_CONSTRUCTOR
	- The controller library is not in a path read by mc_rtc
	- The controller constuctor segfaults
	- The controller library hasn't been properly linked
[critical] No controller enabled
[critical] === Backtrace ===
 0# void mc_rtc::log::error_and_throw<std::runtime_error, char const (&) [22]>(char const (&) [22]) in /usr/local/lib/libmc_control.so.2
 1# mc_control::MCGlobalController::MCGlobalController(mc_control::MCGlobalController::GlobalConfiguration const&) in /usr/local/lib/libmc_control.so.2
 2# mc_control::MCGlobalController::MCGlobalController(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<mc_rbdyn::RobotModule>) in /usr/local/lib/libmc_control.so.2
 3# main in /home/hmr/devel/src/catkin_ws/devel/lib/mc_rtc_ticker/mc_rtc_ticker
 4# __libc_start_main at ../csu/libc-start.c:342
 5# _start in /home/hmr/devel/src/catkin_ws/devel/lib/mc_rtc_ticker/mc_rtc_ticker

terminate called after throwing an instance of 'std::runtime_error'
  what():  No controller enabled
Aborted (core dumped)
```

I would greatly appreciate any assistance in resolving these issues and successfully running the new controller.

Thank you in advance for your help!

gergondet, 2023-09-04:
Hi @Saeed-Mansouri 

There's a few things to check as hinted by the error message

> The controller name does not match the name exported by CONTROLLER_CONSTRUCTOR

For lipm_walking_controller, that's in `src/lib.cpp`. Make sure the name in the `CONTROLLER_CONSTRUCTOR` macro is indeed `XO`

> The controller library is not in a path read by mc_rtc

That would be the case if you did not installed in a destination configured in` ControllerModulePaths` but given the configuration is loaded from `/usr/local/lib/mc_controller/etc` I am going to assume this part is ok (i.e. you have `/usr/local/lib/mc_controller/xo_controller.so``)

> The controller library hasn't been properly linked

This is the most common issue.

As a first step, run: `ldd /usr/local/lib/mc_controller/xo_controller.so`. If some files are not found then you should figure out why.

If that does not work, prefix the command with `LD_DEBUG=libs` (e.g. `LD_DEBUG=libs mc_rtc_ticker 2> ld.log`) then scan the output for errors. Note that some errors are normal (everything with `_LTX_` in it, errors about `LOAD_GLOBAL` and errors about `create_args_required`). Specifically look around the part where the load of `xo_controller.so` is happening.

> The controller constuctor segfaults

That one should probably be removed. It is not caught by mc_rtc anymore.

Saeed-Mansouri, 2023-09-05:
Thank you for your assistance! It's quite interesting that the issue has been resolved by simply shutting down the PC and starting it again. Sometimes, a good old restart does the magic. Now, everything is working well without any changes or rebuilds. Thank you again for your support!

closed 2023-09-05 by Saeed-Mansouri