jrl-umi3218/mc_rtc #390: Error when selecting checkbox for collision monitors in Rviz

mmurooka, 2023-09-10:
In the `Collisions`->`Monitors` tab of Rviz, I get the following error when I click on the checkbox for a collision pair. I don't think it depends on the controller.

```bash
[critical] [CollisionConstraint] Attempted to toggleCollisionMonitor on non-existent collision
terminate called after throwing an instance of 'std::runtime_error'
[critical] === Backtrace ===
 0# void mc_rtc::log::error_and_throw<std::runtime_error, char const (&) [84]>(char const (&) [84]) at /home/mmurooka/workspace/src/mc_rtc/include/mc_rtc/logging.h:52
 1# mc_solver::CollisionsConstraint::toggleCollisionMonitor(int, mc_rbdyn::Collision const*) at /home/mmurooka/workspace/src/mc_rtc/src/mc_solver/CollisionsConstraint.cpp:406
 2# mc_rtc::gui::StateBuilder::ElementStore::ElementStore<mc_rtc::gui::details::CheckboxImpl<mc_solver::CollisionsConstraint::addMonitorButton(int, mc_rbdyn::Collision const&)::{lambda()#1}, mc_solver::CollisionsConstraint::addMonitorButton(int, mc_rbdyn::Collision const&)::{lambda()#2}> >(mc_rtc::gui::details::CheckboxImpl<mc_solver::CollisionsConstraint::addMonitorButton(int, mc_rbdyn::Collision const&)::{lambda()#1}, mc_solver::CollisionsConstraint::addMonitorButton(int, mc_rbdyn::Collision const&)::{lambda()#2}>, mc_rtc::gui::StateBuilder::Category const&, mc_rtc::gui::ElementsStacking, void*)::{lambda(mc_rtc::gui::Element&, mc_rtc::Configuration const&)#4}::_FUN(void*, mc_rtc::Configuration) at /home/mmurooka/workspace/src/mc_rtc/include/mc_rtc/gui/StateBuilder.hpp:129
 3# mc_rtc::gui::StateBuilder::handleRequest(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, mc_rtc::Configuration const&) at /home/mmurooka/workspace/src/mc_rtc/src/mc_rtc/gui/StateBuilder.cpp:236
 4# mc_control::ControllerServer::handle_requests(mc_rtc::gui::StateBuilder&, char const*) at /home/mmurooka/workspace/src/mc_rtc/src/mc_control/ControllerServer.cpp:54
 5# mc_control::ControllerServer::handle_requests(mc_rtc::gui::StateBuilder&) at /home/mmurooka/workspace/src/mc_rtc/src/mc_control/ControllerServer.cpp:90
 6# mc_control::MCGlobalController::run() at /usr/include/c++/9/bits/shared_ptr_base.h:1021
 7# main at /home/mmurooka/workspace/src/catkin_ws/src/mc_rtc_ros/mc_rtc_ticker/src/mc_rtc_ticker.cpp:225
 8# __libc_start_main at ../csu/libc-start.c:342
 9# _start in /home/mmurooka/workspace/src/catkin_ws/devel/lib/mc_rtc_ticker/mc_rtc_ticker

  what():  [CollisionConstraint] Attempted to toggleCollisionMonitor on non-existent collision
```

gergondet, 2023-09-11:
Hi @mmurooka 

Thanks for reporting this. It is fixed in the above commit.

closed 2023-09-11 by gergondet

mmurooka, 2023-09-11:
@gergondet  Thanks, I have confirmed that it has been resolved. When the robot is in motion, the `Monitors` check boxes appear and disappear, making it difficult to check them. I think this is a (low priority) point to be improved.

gergondet, 2023-09-13:
This is because monitoring is not part of the collision description so whenever collisions are removed/added the user interaction that lead to their monitoring is lost. Note that this is not particularly related to the auto-monitoring functionality that was added recently.

Internally, the collision gets a new id in that case too so that's a bit tricky to keep track of collision that are toggled back.

Another option would be to add a monitor toggle (or monitor distance) in the collision description

commit 5cbe59fb (2023-09-11): [mc_solver/CollisionConstraint] Fix crash in monitor activation

Not exiting early after finding the right collision would lead to an
exception being thrown despite the collision existing.

Closes #390