jrl-umi3218/mc_rtc #397: FSM controller in practice, error in loading Transitions

MTdeVries, 2023-10-04:
Hi
Following the tutorial "FSM controller in practice", after creating the controller with the command
mc_rtc_new_fsm_controller my_first_fsm_controller MyFirstFSMController ,
building and running it as specified, I get the following error message which I can't figure out to solve.

"None of the transitions you attempted to load in this TransitionMap are valid." 
The controller is found and read . The confi file ~/.config/mc_rtc/mc_rtc.yaml has the right controller name. So at this moment I'm not sure why all the states are declared empty. Which results in the following error message:

[error] Some states could not be loaded as their base is not available, check for typos or cycles
[warning] - MoveHandle (base: Posture)
[warning] - OpenDoor (base: MoveHandle)
[warning] - OpenDoorDemo (base: Parallel)
[warning] - OpenDoorFSM (base: Meta)
[warning] - ReachHandle (base: MetaTasks)
[warning] - Standing (base: MetaTasks)
[error] Invalid transition:
[error] - origin state (OpenDoorDemo) is not loaded
[error] - destination state (OpenDoorDemo) is not loaded
[critical] None of the transitions you attempted to load in this TransitionMap are valid.
terminate called after throwing an instance of 'std::runtime_error'

Furthermore, I have done all previous tutorials which all worker fine. I did however found a previous bug/mistake which needs mentioning. Namely, the src/MakeList.txt example has the command : add_fsm_controller(mc_door_sample ${controller_SRC} ${controller_HDR}) . However this is no longer a useful command, this is solved by removing the " fsm_"  part.

Hope you can help me get the tutorial working, 

With kind regards,

gergondet, 2023-10-05:
Hi @MTdeVries 

The error message is saying that the states you defined in your yaml file have non existing bases. These bases are the default states provided by mc_rtc.

Maybe your configuration is missing some entries in the `StatesLibraries` section.

The `mc_rtc_new_fsm_controller` command should have two entries:
```
StatesLibraries:
- "@MC_STATES_DEFAULT_RUNTIME_INSTALL_PREFIX@"
- "@MC_STATES_RUNTIME_INSTALL_PREFIX@
```

Which should result (assuming your install prefix is `/usr/local`) into:
```
StatesLibraries:
- "/usr/local/lib/mc_controller/fsm/states"
- "/usr/local/lib/mc_controller/TestFSM/states"
```

MTdeVries, 2023-10-09:
Hi @gergondet 

Unfortunately, this isn't the problem since all states mentioned in the yaml file can be found in configuration files on the location mentioned above. However, creating the FSM project with the specified command also includes a states.json file (located under projectname/src/states/data) which is empty at the moment. Could this impose a problem in the project?

At the moment I'm considering multiple causes which could impose problems for the result. For example, there are some other warning error that I would like to mention, namely : 

Multiple errors mentioning that LOAD_ GLOBAL is an undefined symbol:

[warning] Could not get symbol LOAD_GLOBAL in library /usr/lib/x86_64-linux-gnu/mc_controller/passthrough.so
/usr/lib/x86_64-linux-gnu/mc_controller/passthrough.so: undefined symbol: LOAD_GLOBAL

[info] Robots loaded in controller:
[info] - jvrc1
[info] - ground
[info] - door
[info] MCController(base) ready
[warning] State file /data does not exist
[warning] State file  does not exist
[info] Added task FSM_posture_jvrc1
[info] Added task FSM_posture_door

Above mentioned warnings are posted above the errors listed in the previous message. In the configuration file called DoorSample.in.yaml the location of the statefile was automatically set to the following:

StatesFiles:
- "@MC_STATES_DEFAULT_RUNTIME_INSTALL_PREFIX@/data"
- "@MC_STATES_RUNTIME_INSTALL_PREFIX@/data"
I assumed this wouldn't cause a problem. However you may determine this differently.

Hope by mentioning these warnings, the problem could be resolved.

Kind regards,