jrl-umi3218/mc_rtc #400: What would be required to use mc_rtc with Ubuntu 22?

fjandrad, 2023-10-10:
Hi,
We might be interested in using mc_rtc with ubuntu 22. It seems in the webpage it says mainly ubuntu 18 and 20. 
What would be needed to be used with Ubuntu 22?
Is it just a matter of not being tested yet or is there something not working with it?

@jlee1218

gergondet, 2023-10-10:
Hi @fjandrad 

The documentation should mention 22.04 indeed :)

Since [v2.6.0](https://github.com/jrl-umi3218/mc_rtc/releases/tag/v2.6.0) packages are available for 22.04 and ROS2 iron.

To build from source you can use the deprecated [build-and-install](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/installation-build-and-install.html) method but this does build ROS support into mc_rtc.

I'm working on updating [mc-rtc-superbuild](https://github.com/mc-rtc/mc-rtc-superbuild/) to handle 22.04 and ROS2.

There is also no ROS2 version of [mc_rtc_ros](https://github.com/jrl-umi3218/mc_rtc_ros) or [mc_rtc_ros_control](https://github.com/mc-rtc/mc_rtc_ros_control) for now.

gergondet, 2023-10-16:
> The documentation should mention 22.04 indeed :)

The documentation website had not been updated since April but it should be fixed with https://github.com/jrl-umi3218/mc_rtc/commit/d228037e623d812692f33af2e9dd87a782b1722d

> I'm working on updating [mc-rtc-superbuild](https://github.com/mc-rtc/mc-rtc-superbuild/) to handle 22.04 and ROS2.

This is now working correctly. mc-rtc-superbuild will install ROS2 Iron and use colcon to build ROS dependencies.

I have also confirmed the ROS2 plugin works correctly starting from https://github.com/jrl-umi3218/mc_rtc/releases/tag/v2.9.2

I'm going to close this issue but feel free to open a new one if you encounter issues with 22.04 or the framework

closed 2023-10-16 by gergondet

fjandrad, 2023-10-18:
Thank you, I will try again with your suggestions, and let you know if we face further problems.