jrl-umi3218/mc_rtc #407: Issues creating a new robot module

jlee1218, 2023-11-08:
I am currently having issues with building my new robot module. I am using ubuntu 22.04 for this. 

I can successfully build the robot description package, but I can't seem to get the robot module to build because of this error: https://github.com/jrl-umi3218/mc_rtc/blob/3763fef93b3c88a1f370ea6d791209f9d55fd1b1/CMakeLists.txt#L213

Do you have instructions on how to build a robot module / the environment setup needed so that mc_rtc and these packages can be discovered by each other during the cmake process?

For reference, these are my packages: 
      Robot Module: https://github.com/jlee1218/mc_iisy11_r1300
      Robot Description: https://github.com/jlee1218/mc_iisy11_r1300_description

gergondet, 2023-11-10:
Hi @jlee1218 

The `find_description_package` macro used [here](https://github.com/jlee1218/mc_iisy11_r1300/blob/main/src/CMakeLists.txt#L3) is doing two things:
- Call `find_package(mc_iisy11_r1300_description REQUIRED)`
- Look for the `mc_iisy11_r1300_description_INSTALL_PREFIX` or `mc_iisy11_r1300_description_SOURCE_PREFIX` 

This last part relies on catkin behavior that was removed with ament.

If you build your description package as an ament package you need to:
- export an additional CMake file (example: https://github.com/jrl-umi3218/mc_rtc_data/blob/ROS2/jvrc_description/CMakeLists.txt#L36)
- this CMake file must define the `mc_iisy11_r1300_description_INSTALL_PREFIX` variable (example: https://github.com/jrl-umi3218/mc_rtc_data/blob/ROS2/jvrc_description/cmake/jvrc_description_INSTALL_PREFIX.cmake)

If you build it as a pure CMake package you need to generate a config file.

As a quick work-around you can modify the code of your module:
```
if(NOT DEFINED MC_ISSY11_R1300_DESCRIPTION_PATH)
	find_description_package(mc_iisy11_r1300_description)
endif()
```
And pass the `MC_ISSY11_R1300_DESCRIPTION_PATH` when you build your module.

jlee1218, 2023-11-14:
Thank you very much! This worked perfectly.

closed 2023-11-14 by jlee1218