jrl-umi3218/mc_rtc #437: ISSUE ABOUT  CPU USAGE RATE

EthanZengg, 2024-02-28:
HI, @gergondet 
I attempted to use the  BaselineWalkingController to control a 48-dof robot with mc_rtc_ticker(200hz control loop frequency) . However I observed that the both the calculation speed of mc and the CPU usage rate are  low , are there any solutions available to make full use of CPU?（Or perhaps this is a normal phenomenon?）

[image]

[image]

[image]

hope to receive your answer

gergondet, 2024-02-29:
Hi @EthanZengg 

This is not normal for computation times at least. Have you built the controller with `CMAKE_BUILD_TYPE` set to `RelWithDebInfo` or `Release`?

EthanZengg, 2024-02-29:
Hi, @gergondet,
Thank you for your response. Yes, I built it using the following command.
`catkin build baseline_walking_controller -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_OSQP=ON --catkin-make-args all tests
`

gergondet, 2024-02-29:
I'm not sure this is the correct way to pass the build type, could you try:

```
catkin build baseline_walking_controller --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_OSQP=ON --catkin-make-args all tests 
```

EthanZengg, 2024-02-29:
I thank it should be correct .However I rebuilt it using the new method ,and  the problem still persists. This is result displayed by mc_bin_perf when using the JVRC1

[image]

closed 2024-03-13 by EthanZengg