jrl-umi3218/mc_rtc #453: Request for Code Style Guidelines

Saeed-Mansouri, 2024-06-18:
Could you share the code style guidelines used in the mc_rtc project? 
Thanks!

Saeed-Mansouri, 2024-06-25:
Hello @arntanguy @gergondet,

I hope this message finds you well. Last week, I inquired about the code style guidelines for the `mc_rtc` project but haven't received a response yet. Could you please direct me to where I might find this information, or let me know if it's available? Understanding these guidelines will greatly assist me in contributing properly to the project.

Thank you for your attention to this matter!

Best regards,

arntanguy, 2024-06-26:
Sorry for the delay, Pierre nor myself are no longer actively involved in the project, though we will keep maintaining it as best as possible.

As for style guidelines, we are using [pre-commit](https://pre-commit.com/) to ensure that all style guidelines are enforced.

You can use it as follows (from mc_rtc directory):
```sh
pre-commit run --all-files
````
This will run all the checks and fix any style mistake found.

You can also set-up a git hook that will run these checks for you before commit to ensure that every commit will enforce these rules
```
pre-commit install
````

Besides this, we generally follow the following coding rules:
- Prefer use of `struct` to `class`. This is a somewhat arbitrary preference choice, the only difference is that it allows us to put the public API first without explicitly having to use the `public` keyword.
- Use references instead of pointers wherever suitable
- Respect const correctness
- Pay attention to the return type of lambda functions (for GUI and logging). By default in C++ all lambda functions return by value. For performance reasons it is often preferable to return data by const reference instead (as long as you are sure that the data will continue to exist for at least as long as the callback).
```
auto callback = []() -> const T & { return value; }
````
- Documentation: everything added to the public API should be documented using Doxygen. Big features should be appropriately documented in the tutorial.

That's the main things I can think of on top of my head. Even if we are currently less active, do not hesitate to submit pull requests, and we will be happy to review them.

Best regards,
Arnaud

closed 2024-06-26 by arntanguy

Saeed-Mansouri, 2024-07-08:
Hi @arntanguy,

Thank you for the detailed response and the guidance on using pre-commit to maintain code style. I've also reviewed the configuration files, and I see that they define specific rules covering various aspects of formatting, alignment, and structure for C++ files.

However, these settings seem to apply to a subset of possible style guidelines. For instance, there are detailed rules about column limits, indentation styles, brace breaking, and even the alignment of pointers, but it's not entirely clear if these are derived from a broader, recognized style guide.

Could you please clarify if these configurations are based on a standard style guide, such as the Google Code Style, or if they are custom guidelines developed specifically for the mc_rtc project? Understanding this would help me align my contributions more closely with the project's coding conventions. Thanks!

arntanguy, 2025-03-14:
> Understanding this would help me align my contributions more closely with the project's coding conventions.

I'm not sure what you mean by that, the whole point of `pre-commit` is that all these conventions are automatically followed. If in fact you push something that does not match these styling conventions, the CI will fix it for you. Coding style is loosely based on `Alllman` with a few specific rules to make constructs that `mc_rtc` uses heavily look "nice" (i.e lamdas used for gui/logging).

As for deeper C++ rules we don't strictly enforce a standard such as google code style. We go with the pragmatic write-sensible code approach. Be mindful of performance and readability.

 One thing though, we are particularly attached to API stability and backward compatibility (at least for minor versions). Thus breaking changes should be avoided as much as possible, especially when they are runtime behavior changes. It's best wherever possible when changing how something is done (for example loading from YAML) to mark the old feature as deprecated, and implement an automatic way to make the old code keep working.