jrl-umi3218/mc_rtc #470: [critical] boundedSpeed is already handled by another loading function

Geson456, 2025-05-15:
I'm new to the mc_rtc control framework. I installed mc_rtc using the ./build_and_install.sh script in mc_rtc/utils, and also installed the mc_mujoco folder. Now this issue suddenly appeared when I run the command "mc_mujco", and I don’t know the cause. I hope you can provide assistance. Thank you!

[image]

[image] @gergondet

arntanguy, 2025-05-15:
The `build_and_install.sh` script is deprecated, the recommended way to build the framework is to use https://github.com/mc-rtc/mc-rtc-superbuild

Regarding your issue, the most likely cause is that you have several versions of mc_rtc installed in the same `LD_LIBRARY_PATH`. Can you check

```
ls <install_prefix>/lib/libmc*
```

If you have multiple `.so` versions for each library, remove the old ones.

Geson456, 2025-05-16:
Thank you for pointing that out. There were indeed many ​​libmc_*_d.so​​ files, which are likely debug versions of the libraries. The issue was resolved after deleting them.

closed 2025-05-19 by arntanguy